carla-simulator/carla #2213: road::InformationSet::GetInfos() not working for RoadInfoSpeed

amastrobera, 2019-11-06:
Dear Carla devs, 

I am using the code in [here](https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/road/Road.cpp) line 205: 

```auto geom_info_list = _info.GetInfos<element::RoadInfoGeometry>();```

I wanted to do the same thing for RoadInfoSpeed, like `_info.GetInfos<element::RoadInfoSpeed>();` but I get this error: 

```
../road/element/RoadInfoIterator.h:58:36
error:  invalid static_cast from type 'carla::road::element::RoadInfo' to 
            type 'carla::road::element::RoadInfoSpeed&' 
                        return static_cast<T &>(**_it);
```

Do you know what's wrong ?

amastrobera, 2019-11-06:
my error :) nevermind

closed 2019-11-06 by amastrobera

tRNAoO, 2024-01-18:
Hi, I get the same error when using `map.get().GetLane(wpt).GetRoad()->GetInfo<RoadInfoLaneWidth>(0.0);`  and ` map.get().GetLane(wpt).GetRoad()->GetInfo<RoadInfoSpeed>(0.0);` :
```
error: invalid static_cast from type ‘carla::road::element::RoadInfo’ to type ‘carla::road::element::RoadInfoLaneWidth&’
       return static_cast<T &>(**_it);
```
Can you remember how to solve this problem? Thanks!

tRNAoO, 2024-01-19:
Solved, just `#include <carla/road/element/RoadInfoSpeed.h>` :)