carla-simulator/carla #2227: Is there already a data collector implementation for Carla 0.9.6 ?

aakashsehgal, 2019-11-14:
I want to try end to end deep learning using Carla. For that I require the images as well as the corressponding steering angle for each image. I would drive manually and record this data.Is there already one such implementation or a guidance of how to do it in Carla 0.9.6 ?

I would want to save it in csv file as shown:

[image]

cassfalg, 2019-11-14:
Take a look at tutorial.py, should be a start, though you may want to modify manual_control.py to write your csv file and images. It'll require some python coding.

AlanNaoto, 2019-11-26:
I'm working in one for RGB, bounding box and depth data for my master thesis. Later I plan on sharing it with the community :)

aakashsehgal, 2019-11-26:
I already built one customized version using pandas library. I added this functionality in the manual_control.py

e270889o, 2019-12-02:
> I already built one customized version using pandas library. I added this functionality in the manual_control.py

Would you share your implementation?

closed 2020-02-13 by stale[bot]

VEDANTGHODKE, 2020-10-11:
@aakashsehgal Could you please share your implementation?