carla-simulator/carla #2230: A very simple controller?

ndeshp2s, 2019-11-15:
Is there an availability of a very simple controller in CARLA? I have high level actions as: accelerate, decelerate, continue, stop.  So for say accelerate, I want to increase the current velocity of the ego vehicle?

I tried the PID controller implementation of the BasicAgent. But its very noisy. I tried set_velocity() function but then there is problem while turning. I have a list of waypoints to be followed. I was wondering if there something already available to find the the correct value for   set_angular_velocity()? based on next waypoint? 

Or if there is some other way I could get the control working?

jackbart94, 2019-11-20:
You can tune the PID controller in the BasicAgent, or you can try with either the RoamingAgent, or the BehaviorAgent (see pull requests).

ndeshp2s, 2019-12-05:
Yes @jackbart94. I have been tuning the existing PID controller. However, there seems some issue with braking. Even if I give any values between 0.1 to 1, it  always applies 1 to the brake parameter.

closed 2020-02-10 by stale[bot]