carla-simulator/carla #2231: recipe for target 'launch' failed

jingbaixue, 2019-11-16:
Hi
when i make launch, I got the error as belows:
[2019.11.16-05.22.08:503][  0]LogSlate: New Slate User Created.  User Index 0, Is Virtual User: 0
[2019.11.16-05.22.08:503][  0]LogSlate: Slate User Registered.  User Index 0, Is Virtual User: 0
[2019.11.16-05.22.09:177][  0]LogInit: Using SDL_WINDOW_VULKAN
[2019.11.16-05.22.09:177][  0]LogVulkanRHI: Display: - Found instance extension VK_EXT_debug_report
[2019.11.16-05.22.09:177][  0]LogVulkanRHI: Display: Not using instance layers
[2019.11.16-05.22.09:177][  0]LogVulkanRHI: Display: Not using instance extensions
[2019.11.16-05.33.10:212][  0]LogCore: FUnixPlatformMisc::RequestExit(bForce=true, ReturnCode=1)
[2019.11.16-05.33.10:212][  0]LogCore: FUnixPlatformMisc::RequestExit(1)
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 1

and 
[image]

what should i do

cassfalg, 2019-11-18:
Google "vulcan driver", understand what it is, install it? ;)

Make sure that you have a supported GPU and that you have Vulcan drivers installed for said GPU. The Server itself you can start with the `-opengl` flag. The editor too, probably, but I don't know how.

jingbaixue, 2019-11-19:
@cassfalg thank you for your help, i found my computer doesn't have the graphics card it needs

closed 2019-11-19 by jingbaixue