carla-simulator/carla #2232: Invalid pixelmap or window parameter

rubentfd, 2019-11-18:
Hi!
I'm using 0.9.6 and I always get the same error:
Disabling core dumps.
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x3e00041
  Serial number of failed request:  361
  Current serial number in output stream:  367
Signal 11 caught.
Segmentation fault (core dumped)

I don't know what caused this problem.

cassfalg, 2019-11-18:
Please provide information about your OS, GPU and GPU drivers.

rubentfd, 2019-11-18:
Ubuntu 18.04 and I have an Nvidia GeForce MX250 with 440.26 driver installed

cassfalg, 2019-11-19:
Hm... What have you tried already to fix/investigate this?

rubentfd, 2019-11-19:
I dodn't trie anything, because I'm not an expert in this and I don't wanted to mess up something.

cassfalg, 2019-11-20:
I'm afraid I can't help you further then because I'm no expert either and everything I can do at this point (like googling the error) you can do as well.

mahmoudhamdy, 2019-12-26:
did you succeed to find a solution for this issue?, I have the same problem.

Ubuntu 18.04, GeForce GTX 1050 Ti, driver 440. 

I think it might be drivers issue

rubentfd, 2019-12-26:
I switched to an older version and it worked. Try to change drivers in: Software & updates/Additional Drivers. Might gonna work

mahmoudhamdy, 2019-12-26:
Which version did u switch to?  .. which drivers did you change? of nvidia? the problem i have 440 drivers and i had several problems getting ubuntu to work with nvidia gtx 1050 with other versions of drivers like 435

epoch-11, 2019-12-28:
confirmed same error w/ NVidia 440 driver
Ubuntu 18.04
GTX150
Not sure if my GTX150 is going to cut it on my laptop anyway. Downgrading is challenging b/c of all the other components.

mahmoudhamdy, 2019-12-29:
nvidia-driver-390 is working with 0.9.6 I think, but I am not sure if it will work with Ubuntu 18.04 as I always had logging issues because of the older version of the drivers, I was always stuck after the window login.

najibghadri, 2020-02-19:
@rubentfd Which version worked for you?

rubentfd, 2020-02-19:
0.9.3

najibghadri, 2020-02-19:
> 0.9.3
:'D
We already know that, the question was which nvidia driver..

> Which version did u switch to? .. which drivers did you change? of nvidia? the problem i have 440 drivers and i had several problems getting ubuntu to work with nvidia gtx 1050 with other versions of drivers like 435

rubentfd, 2020-02-23:
Oh right sorry. Currently I am using 440.48.02

jediofgever, 2020-03-16:
This issue is still persistent with newest versions of Carla and NVIDIA

closed 2020-05-22 by stale[bot]

varunjammula, 2020-08-19:
Same issue with Ubuntu 18, Driver Version: 450.51.06    CUDA Version: 11.0  

It works with opengl though!

adityapatruni, 2020-10-21:
> Same issue with Ubuntu 18, Driver Version: 450.51.06 CUDA Version: 11.0
> 
> It works with opengl though!

Which version of OpenGL? And what should I do after I install OpenGL? Thanks.

varunjammula, 2020-10-25:
https://carla.readthedocs.io/en/latest/adv_rendering_options/

KoulickS, 2020-11-19:
I am getting something like this:
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  130 (MIT-SHM)
  Minor opcode of failed request:  3 (X_ShmPutImage)
  Resource id in failed request:  0x400000f
  Serial number of failed request:  1863
  Current serial number in output stream:  1864

Ubuntu 18.04, Integrated Intel graphics. Any solution how to fix this?

manju838, 2021-03-17:
I am getting the same error like this:

INTEL-MESA: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x3000041
  Serial number of failed request:  370
  Current serial number in output stream:  381
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)

I am using Ubuntu 20.04 dual boot laptop with Nvidia MX230 GPU[2GB] ,driver version 460.39 and RAM 20 GB.I faced the same problem with Ubuntu 18.04 before upgrading to Ubuntu 20.04.I tried Graphics Rendering previously but to no use.
Please suggest a fix,I am facing this issue since December 2020

leeyuanalibaba, 2021-04-25:
same issue,  
4.26.1-0+++UE4+Release-4.26 522 0
Disabling core dumps.
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x680003e
  Serial number of failed request:  399
  Current serial number in output stream:  409
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)

issue related with nvidia driver(my env Kubuntu 20.10 cuda 10.2, driver ver 460 installed through by apt,  Geforce RTX 2060 Super),

dgenin, 2021-06-09:
Same issue:
```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
INTEL-MESA: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x2e00041
  Serial number of failed request:  398
  Current serial number in output stream:  409
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)
```
Doesn't seem to be stale at all.

harshnehal1996, 2021-06-23:
Using -opengl option worked for me as was mentioned in similar thread

heethesh, 2021-08-13:
Same issue with 0.9.12 (Driver Version: 460.91.03, CUDA Version: 11.2), seems like `-opengl` is no longer supported now. I'm not even able to run through the Docker container in 0.9.12.

jbeck9, 2021-08-20:
> Same issue with 0.9.12 (Driver Version: 460.91.03, CUDA Version: 11.2), seems like `-opengl` is no longer supported now. I'm not even able to run through the Docker container in 0.9.12.

I'm in the same boat. I've been avoiding this by running CARLA with opengl. Now that opengl support is gone I am stuck. I've tried to remedy this with multiple different driver versions with no success.

heethesh, 2021-08-20:
@jbeck9 I actually got this working by switching to the latest Nvidia driver 470 and running Carla with the -RenderOffScreen flag. This directly worked outside the Docker environment itself.

jbeck9, 2021-08-20:
> @jbeck9 I actually got this working by switching to the latest Nvidia driver 470 and running Carla with the -RenderOffScreen flag. This directly worked outside the Docker environment itself.

Thank you for the quick reply. It turns out the root problem for me was that Vulkan was using Intel graphics by default instead of my NVIDIA GPU. No idea why.

For anyone who is getting errors preceded by:

`INTEL-MESA: <Error or warning>`

Even though you have an NVIDIA GPU that is active. Run this:

`export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"`

Carla should then run normally with Vulkan.

`./CarlaUE4.sh`

mvalentin123, 2021-08-25:
The nvidia_icd.json file was missing in my case. Had to create one like this one
`
{
    "file_format_version" : "1.0.0",
    "ICD": {
        "library_path": "libGLX_nvidia.so.0",
        "api_version" : "1.2.131"
    }
}
`

But now Carla 0.9.12 works. Thanks @jbeck9

matejm42, 2023-05-08:
The issue still persist.

```
./CarlaUE4.sh 
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Serial number of failed request:  299
  Current serial number in output stream:  309
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)
```

Carla 0.9.13
Ubuntu 20.04 
Nvidia GeForce RTX 2060
nvidia-driver-510

```./CarlaUE4.sh -opengl```

> Warning:  OpenGL is no longer supported for desktop platforms. Vulkan will be used instead.
and crashes

```
export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"  && ./CarlaUE4.sh
```
fixes the issue

yhojann-cl, 2023-08-01:
Same issue on Ubuntu 22.04 LTS and AMD Radeon RX 6800 XT card.