carla-simulator/carla #2236: Low RGB Camera sensor quality

rajatsaini0294, 2019-11-21:
Hi Carla team,

I am using the Carla 0.9.5. The RGB camera sensor quality is not good [images attached]. Can you guys suggest any ways to improve the camera sensor quality?

[image]
[image]
[image]
[image]

cassfalg, 2019-11-22:
What do you mean by "pixelated"?

rajatsaini0294, 2019-11-22:
The objects in the camera sensor image appears to be very far. Also the objects are not clear enough. I was expecting much better quality images from the camera sensor.

cassfalg, 2019-11-22:
Better image quality based on what? What settings do you use? Did you disable postprocessing effects for example?

AlanNaoto, 2019-11-22:
Hey @rajatsaini0294, I am on 0.9.6 and had a similar first impression when comparing the native ./CarlaUE4.sh viewer. What I did was increase the rgb camera sensor resolution up to 1920x1080, and then it seemed better. Though for some weather settings when there are too many clouds / not too much sun lighting then the image appears similar to yours, so I guess light plays a big role in real life fidelity :)

Here is a sample of my (yet incomplete) capture set: https://www.youtube.com/watch?v=aTqnrflBRWU

cassfalg, 2019-11-22:
Just compare images of this weather to some of the rainy weathers where you have reflections from wet surfaces. They look way better. Or use the default weather on Town03 (or some sunset weather) where you have lots of shadows. They look a lot better.

IMHO as far as "realistic" image quality is concerned, Carla would really benefit from better (as in more diverse, higher resolution) textures, and to some extend from more complex assets (buildings, walls, trees...). That's why the wet conditions look a lot better in my opinion - the reflections provide more "interesting and diverse" textures. So do shadows. This weather condition has neither, hence the "bland" look.

sukritkalra, 2019-11-22:
@AlanNaoto Thanks for sharing the video with the 2D bounding boxes. Can you elaborate on how you tackled occlusion to get the bounding boxes for the vehicles and the pedestrians?

AlanNaoto, 2019-11-22:
> @AlanNaoto Thanks for sharing the video with the 2D bounding boxes. Can you elaborate on how you tackled occlusion to get the bounding boxes for the vehicles and the pedestrians?

Hey there. I check if the proposed bbox coordinate has a Z value lower or equal than the same pixel position on the depth map. If so, then it means that this point is shown in the depth map and thus not occluded. If not, then that point is occluded!

rajatsaini0294, 2019-11-25:
Thank you all for suggestions. Changing the camera resolution to 1920*1080 and some weather settings increases the camera sensor quality.
One more thing: If i increase the FPS of the CARLA, the simulation becomes very slow. Is there any way to tackle this thing?

AlanNaoto, 2019-11-25:
You can try checking the example script "synchronous_mode.py" 😊

closed 2019-12-04 by rajatsaini0294