carla-simulator/carla #2237: Possbility to use OpenDrive map only without fbx data to generate Carla maps

hanqo, 2019-11-21:
Currently I am using Carla for some planning/controlling with feedback loop simulation, and it works well with default maps. However, when I want to try some of my self defined OpenDrive maps, it seems that I need to use many tools like RoadRunner together with UE4 Editor to generate a carla map. That is usually too much work to generate one map for my single test case. I am not so interested into mesh data because I simulate only with high level planning. So is there a easy solution to use only my OpenDrive map with e.g. Hero Mode running on Carla Server?

LawranceXcl, 2019-12-03:
Good question! I recently encountered the same problem. The process of creating my own map is complicated so much by RoadRunner together with UE4. I really hope that carla perfects the map function as soon as possible.

daydin41, 2020-02-04:
I am facing the same problem. Is there anything carla team can suggest?

closed 2020-04-11 by stale[bot]