carla-simulator/carla #2248: CARLA crashes because of AssertionError in TireConfig.cpp

Ajk4, 2019-11-27:
Hi.

I'm using server in synchronous mode to roll sequence of episodes.

I noticed that after some time CARLAs crash with AssertionError from PhysXVehicles. Is it possible that 

```
[2019.11.26-19.00.15:004][649]LogTemp: Loaded OpenDrive file '../../../CarlaUE4/Content/Carla/Maps/OpenDrive/Town03.xodr'
[2019.11.26-19.00.15:071][649]LogTemp: Loaded OpenDrive file '../../../CarlaUE4/Content/Carla/Maps/OpenDrive/Town03.xodr'
[2019.11.26-19.00.15:130][649]LogOutputDevice: Warning: 

Script Stack (0 frames):

Assertion failed: AllTireConfigs[TireConfigID] == this [File:/data/carla/UnrealEngine_4.22/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp] [Line: 89] 

[2019.11.26-19.00.15:134][649]LogCore: Error: appError called: Assertion failed: AllTireConfigs[TireConfigID] == this [File:/data/carla/UnrealEngine_4.22/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp] [Line: 89] 

Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=95424 LargeMemoryPoolOffset=226560 
[2019.11.26-19.00.15:141][649]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000007c78

[2019.11.26-19.00.15:142][649]LogCore: Assertion failed: AllTireConfigs[TireConfigID] == this [File:/data/carla/UnrealEngine_4.22/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp] [Line: 89] 

0x0000000002d27404 CarlaUE4!FUnixErrorOutputDevice::Serialize(char16_t const*, ELogVerbosity::Type, FName const&) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Core/Private/Unix/UnixErrorOutputDevice.cpp:70]
0x0000000002ba96bb CarlaUE4!FOutputDevice::LogfImpl(char16_t const*, ...) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:71]
0x0000000002b497a7 CarlaUE4!FDebug::AssertFailed(char const*, char const*, int, char16_t const*, ...) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:440]
0x0000000002b495cd CarlaUE4!FDebug::CheckVerifyFailedImpl(char const*, char const*, int, char16_t const*, ...) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:416]
0x00000000025aaf6c CarlaUE4!UTireConfig::BeginDestroy() [/data/carla/UnrealEngine_4.22/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp:89]
0x0000000002eefe58 CarlaUE4!UObject::ConditionalBeginDestroy() [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/CoreUObject/Private/UObject/Obj.cpp:942]
0x0000000002e9740e CarlaUE4!UnhashUnreachableObjects(bool, float) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/CoreUObject/Private/UObject/GarbageCollection.cpp:1633]
0x0000000002e96256 CarlaUE4!IncrementalPurgeGarbage(bool, float) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/CoreUObject/Private/UObject/GarbageCollection.cpp:1052]
0x0000000005491855 CarlaUE4!UEngine::ConditionalCollectGarbage() [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/UnrealEngine.cpp:1289]
0x0000000004e5e989 CarlaUE4!UWorld::Tick(ELevelTick, float) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/LevelTick.cpp:1755]
0x0000000004ce634f CarlaUE4!UGameEngine::Tick(float, bool) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/GameEngine.cpp:1360]
0x00000000024b064b CarlaUE4!FEngineLoop::Tick() [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3961]
0x00000000024b6f5a CarlaUE4!GuardedMain(char16_t const*) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Launch/Private/Launch.cpp:170]
0x0000000005bf863f CarlaUE4!CommonUnixMain(int, char**, int (*)(char16_t const*)) [/data/carla/UnrealEngine_4.22/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:243]
0x00002b44bca9d3d5 libc.so.6!__libc_start_main(+0xf4)
0x00000000024a6029 CarlaUE4!_start()

[2019.11.26-19.00.15:152][649]LogExit: Executing StaticShutdownAfterError
[2019.11.26-19.00.15:162][649]LogCore: Warning: Unable to statfs('/memfs/17105058/carla-20191010-096-updated_carla-dev/CarlaUE4/Saved/Crashes/crashinfo-CarlaUE4-pid-31864-175E4FB508D772AB410B3C7E1ACB854E/CarlaUE4.log'): errno=2 (No such file or directory)
Malloc Size=1048576 LargeMemoryPoolOffset=1275152 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
/memfs/17105058/carla-20191010-096-updated_carla-dev/CarlaUE4.sh: line 5: 31864 Segmentation fault      "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4" CarlaUE4 $@
```

This might be connected to https://github.com/carla-simulator/carla/issues/2239 as I noticed that usually client gets frozen on world.tick after CARLA crashes with said AssertionError.

closed 2020-02-02 by stale[bot]

stavBodik, 2020-03-18:
@Ajk4 do you have any news on this issue ?

Ajk4, 2020-04-03:
@stavBodik 
No. I was not able to fix it.

 I recently started doing experiments with many cars (>50) and this issue make it very painful.

Hi, @nsubiron could we reopen this issue? It's still present in CARLA.

walzimmer, 2020-10-03:
I can confirm this error message. It occurs when spawning a new vehicle with a custom 3D model.
I followed this guideline: https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/

```
Caught signal

libUE4Editor-PhysXVehicles.so!UTireConfig::SetFrictionScale(float) [../UnrealEngine-4.24/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp:22]
libUE4Editor-Carla.so!ACarlaWheeledVehicle::SetWheelsFrictionScale(TArray<float, TSizedDefaultAllocator<32> >&) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp:209]
libUE4Editor-Carla.so!ACarlaWheeledVehicle::BeginPlay() [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp:50]
libUE4Editor-Engine.so!AActor::DispatchBeginPlay(bool) [../UnrealEngine-4.24/Engine/Source/Runtime/Engine/Private/Actor.cpp:3382]
libUE4Editor-Engine.so!AActor::PostActorConstruction() [../UnrealEngine-4.24/Engine/Source/Runtime/Engine/Private/Actor.cpp:3225]
libUE4Editor-Engine.so!AActor::FinishSpawning(FTransform const&, bool, FComponentInstanceDataCache const*) [../UnrealEngine-4.24/Engine/Source/Runtime/Engine/Private/Actor.cpp:3124]
libUE4Editor-Engine.so!UGameplayStatics::FinishSpawningActor(AActor*, FTransform const&) [../UnrealEngine-4.24/Engine/Source/Runtime/Engine/Private/GameplayStatics.cpp:689]
libUE4Editor-Engine.so!UGameplayStatics::execFinishSpawningActor(UObject*, FFrame&, void*) [../UnrealEngine-4.24/Engine/Source/Runtime/Engine/Classes/Kismet/GameplayStatics.h:49]
libUE4Editor-CoreUObject.so!UObject::execCallMathFunction(UObject*, FFrame&, void*) [../UnrealEngine-4.24/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:860]
libUE4Editor-CoreUObject.so!UObject::execLetObj(UObject*, FFrame&, void*) [../UnrealEngine-4.24/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:2611]
libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [../UnrealEngine-4.24/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1024]
libUE4Editor-CoreUObject.so!UFunction::Invoke(UObject*, FFrame&, void*) [../UnrealEngine-4.24/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:4907]
libUE4Editor-CoreUObject.so!UObject::ProcessEvent(UFunction*, void*) [../UnrealEngine-4.24/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1910]
libUE4Editor-Engine.so!AActor::ProcessEvent(UFunction*, void*) [../UnrealEngine-4.24/Engine/Source/Runtime/Engine/Private/Actor.cpp:795]
libUE4Editor-Carla.so!ACarlaActorFactoryBlueprint::SpawnActor(FTransform const&, FActorDescription const&, FActorSpawnResult&) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/CarlaActorFactoryBlueprint.gen.cpp:41]
libUE4Editor-Carla.so!ACarlaActorFactoryBlueprint::SpawnActor(FTransform const&, FActorDescription const&) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/CarlaActorFactoryBlueprint.h:37]
libUE4Editor-Carla.so!UActorDispatcher::SpawnActor(FTransform const&, FActorDescription, unsigned int) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDispatcher.cpp:54]
libUE4Editor-Carla.so!UCarlaEpisode::SpawnActorWithInfo(FTransform const&, FActorDescription, unsigned int) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.h:198]
libUE4Editor-Carla.so!FCarlaServer::FPimpl::BindActions()::$_45::operator()(carla::rpc::ActorDescription, carla::geom::Transform const&) const [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Server/CarlaServer.cpp:360]
libUE4Editor-Carla.so! [../UnrealEngine-4.24/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/future:1821]
libUE4Editor-Carla.so!std::__1::packaged_task<carla::rpc::Response<carla::rpc::Actor> ()>::operator()() [../UnrealEngine-4.24/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/future:2089]
libUE4Editor-Carla.so!boost::asio::detail::completion_handler<carla::detail::MoveWrapper<std::__1::packaged_task<carla::rpc::Response<carla::rpc::Actor> ()> > >::do_complete(void*, boost::asio::detail::scheduler_operation*, boost::system::error_code const&, unsigned long) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/detail/completion_handler.hpp:70]
libUE4Editor-Carla.so!boost::asio::detail::scheduler::do_wait_one(boost::asio::detail::conditionally_enabled_mutex::scoped_lock&, boost::asio::detail::scheduler_thread_info&, long, boost::system::error_code const&) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/detail/impl/scheduler.ipp:527]
libUE4Editor-Carla.so!boost::asio::detail::scheduler::wait_one(long, boost::system::error_code&) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/detail/impl/scheduler.ipp:239]
libUE4Editor-Carla.so!unsigned long boost::asio::io_context::run_one_until<std::__1::chrono::steady_clock, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l> > >(std::__1::chrono::time_point<std::__1::chrono::steady_clock, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l> > > const&) [../carla-0.9.9.4-source/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/impl/io_context.hpp:115]
[... cut ...]
libUE4Editor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [../UnrealEngine-4.24/Engine/Source/Editor/UnrealEd/Private/UnrealEdEngine.cpp:407]
UE4Editor!FEngineLoop::Tick() [../UnrealEngine-4.24/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4479]
UE4Editor!GuardedMain(char16_t const*) [../UnrealEngine-4.24/Engine/Source/Runtime/Launch/Private/Launch.cpp:175]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [../UnrealEngine-4.24/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xef)
UE4Editor!_start()
```

darld, 2021-05-13:
Up post Same issue with @walzimmer

rambodrahmani, 2021-10-31:
Any news regarding this? Same issue here. This happens if I try to change parameters in the wheel blueprint (any parameter).

```
[2021.10.31-09.28.08:395][165]LogCore: 0x00007fb066bb4d50 libUE4Editor-PhysXVehicles.so!UTireConfig::SetFrictionScale(float) [/home/rr/DevOps/UnrealEngine/Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/TireConfig.cpp:24]
0x00007fb03bd891a4 libUE4Editor-Carla.so!ACarlaWheeledVehicle::SetWheelsFrictionScale(TArray<float, TSizedDefaultAllocator<32> >&) [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp:250]
0x00007fb03bd88d62 libUE4Editor-Carla.so!ACarlaWheeledVehicle::BeginPlay() [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp:92]
0x00007fb3d38ff620 libUE4Editor-Engine.so!AActor::DispatchBeginPlay(bool) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/Engine/Private/Actor.cpp:3515]
0x00007fb3d38fe133 libUE4Editor-Engine.so!AActor::PostActorConstruction() [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/Engine/Private/Actor.cpp:3347]
0x00007fb3d38fac46 libUE4Editor-Engine.so!AActor::FinishSpawning(FTransform const&, bool, FComponentInstanceDataCache const*) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/Engine/Private/Actor.cpp:3246]
0x00007fb3d411e668 libUE4Editor-Engine.so!UGameplayStatics::FinishSpawningActor(AActor*, FTransform const&) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/Engine/Private/GameplayStatics.cpp:693]
0x00007fb3d516e6c0 libUE4Editor-Engine.so!UGameplayStatics::execFinishSpawningActor(UObject*, FFrame&, void*) [/home/rr/DevOps/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Engine/GameplayStatics.gen.cpp:1386]
0x00007fb3d67aee12 libUE4Editor-CoreUObject.so!UObject::execCallMathFunction(UObject*, FFrame&, void*) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:930]
0x00007fb3d67bdd22 libUE4Editor-CoreUObject.so!UObject::execLetObj(UObject*, FFrame&, void*) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:2689]
0x00007fb3d67b151d libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]
0x00007fb3d653201c libUE4Editor-CoreUObject.so!UFunction::Invoke(UObject*, FFrame&, void*) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:5588]
0x00007fb3d67b7db3 libUE4Editor-CoreUObject.so!UObject::ProcessEvent(UFunction*, void*) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1990]
0x00007fb3d38e8543 libUE4Editor-Engine.so!AActor::ProcessEvent(UFunction*, void*) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/Engine/Private/Actor.cpp:864]
0x00007fb03bd9accf libUE4Editor-Carla.so!ACarlaActorFactoryBlueprint::SpawnActor(FTransform const&, FActorDescription const&, FActorSpawnResult&) [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/CarlaActorFactoryBlueprint.gen.cpp:39]
0x00007fb03bd9c0fc libUE4Editor-Carla.so!ACarlaActorFactoryBlueprint::SpawnActor(FTransform const&, FActorDescription const&) [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/CarlaActorFactoryBlueprint.h:37]
0x00007fb03bbc8123 libUE4Editor-Carla.so!UActorDispatcher::SpawnActor(FTransform const&, FActorDescription, unsigned int) [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDispatcher.cpp:56]
0x00007fb03bbfd0e8 libUE4Editor-Carla.so!UCarlaEpisode::SpawnActorWithInfo(FTransform const&, FActorDescription, unsigned int) [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.cpp:407]
0x00007fb03bcec125 libUE4Editor-Carla.so!FCarlaServer::FPimpl::BindActions()::$_55::operator()(carla::rpc::ActorDescription, carla::geom::Transform const&) const [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Server/CarlaServer.cpp:559]
0x00007fb03bcebf09 libUE4Editor-Carla.so! [/home/rr/DevOps/UnrealEngine/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/future:1816]
0x00007fb03bd48eee libUE4Editor-Carla.so!std::__1::packaged_task<carla::rpc::Response<carla::rpc::Actor> ()>::operator()() [/home/rr/DevOps/UnrealEngine/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/future:2084]
0x00007fb03bd48e13 libUE4Editor-Carla.so!void boost::asio::system_executor::dispatch<carla::detail::MoveWrapper<std::__1::packaged_task<carla::rpc::Response<carla::rpc::Actor> ()> >, std::__1::allocator<void> >(carla::detail::MoveWrapper<std::__1::packaged_task<carla::rpc::Response<carla::rpc::Actor> ()> >&&, std::__1::allocator<void> const&) const [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/impl/system_executor.hpp:39]
0x00007fb03bd48c58 libUE4Editor-Carla.so!boost::asio::detail::executor_op<boost::asio::detail::work_dispatcher<carla::detail::MoveWrapper<std::__1::packaged_task<carla::rpc::Response<carla::rpc::Actor> ()> > >, std::__1::allocator<void>, boost::asio::detail::scheduler_operation>::do_complete(void*, boost::asio::detail::scheduler_operation*, boost::system::error_code const&, unsigned long) [/home/rr/DevOps/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/detail/executor_op.hpp:70]
[... cut ...]
0x000000000026c461 UE4Editor!FEngineLoop::Tick() [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4830]
0x000000000027ad23 UE4Editor!GuardedMain(char16_t const*) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
0x00007fb3cddff286 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/rr/DevOps/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
0x00007fb3cdc250b3 libc.so.6!__libc_start_main(+0xf2)
0x0000000000263d29 UE4Editor!_start()
```

momala, 2022-10-12:
I also have the same issue in Carla 0.9.13. I tried to change the suspension parameters for my custom-built car. It crashes with the same error. Is there any update or idea where it comes from?

momala, 2022-11-02:
> I also have the same issue in Carla 0.9.13. I tried to change the suspension parameters for my custom-built car. It crashes with the same error. Is there any update or idea where it comes from?

I think I've found the problem. After I change any of the parameters in the wheel blueprints and compiled and saved them, I found that the tire configuration became not set. So reconfiguring the Tire config and then saving the blueprint can solve the issue. I hope you'll find it useful.

joschi27, 2024-03-13:
For me, i was missing the TireConfig in the Blueprint of each assigned wheel:
[image]