carla-simulator/carla #2255: Adding new weather conditions to carla

aadityasp1, 2019-12-02:
Hi, I was searching how to add new weather conditions to carla, and I came across this comment. https://github.com/carla-simulator/carla/issues/487#issuecomment-396179843  . However, I am unable to find the mentioned "CarlaWeather.ini " file in the Carla project (Carla 0.9.5) which was built from source. Also, I could not find any option to change the weather and add the presets to a file in Unreal Engine too (as shown in the screenshot of issue #487 ). But, I found this "CarlaWeather.ini" file in the Compiled version of Carla 0.9.5 ( directly downloaded from [here](https://github.com/carla-simulator/carla/releases)).
It would be great if anyone can help me out in understanding the procedure to add new weather conditions in carla. Thanks.

debnera, 2019-12-02:
You can use the Python API for changing the weather, as mentioned in the documentation. This is how it works in 0.9.6 and I believe 0.9.5 is no different.

https://carla.readthedocs.io/en/latest/python_api_tutorial/#changing-the-weather
https://carla.readthedocs.io/en/latest/python_api/#carla.WeatherParameters

aadityasp1, 2019-12-02:
Thanks for the reply @debnera  , I understood how to change the weather on the python client end. However, I have 2 quesions, 
1) I want to understand how can we add new weather conditions like Snow, foggy weather etc. ? 
2) When a value, say precipitation changes on client end (by using set_weather() function) , what is the corresponding value which changes in the Server end? For example, when I run carla simulator from Unreal Engine, and connect a python client to it and change weather in the client, I see the weather change in server window too, but I'm unable to see any variable value change in the Weather Blueprint. Do blueprint variables dynamically change w.r.t changes on the client end?

debnera, 2019-12-03:
I'm not the right person to answer those questions, but I will give it a go anyway.

1. If the desired weather features do not already exist in CARLA, you need to create them yourself in the Unreal Editor. This is more of an Unreal Engine specific question and does not have much to do with CARLA. You could try creating new particle effects and/or using pre-existing visual effects in found Unreal Engine. After this, you may need to add your new effects to the c++ code of CARLA Weather. 
https://docs.unrealengine.com/en-US/Resources/Showcases/Effects/SnowExamples/index.html 
https://docs.unrealengine.com/en-US/Engine/Actors/FogEffects/AtmosphericFog/index.html

2. As far as I know, the settings should be visible somewhere in the editor, but I have never checked it myself. You might be looking at the wrong blueprint. I don't have access to CARLA right now so I can't check it myself. Perhaps you can look at the source code of CARLA Weather to see how and where the values are changed?

closed 2020-02-08 by stale[bot]