carla-simulator/carla #2257: "failed to generate map" when trying to load copied map via PythonAPI

johnnyyuma, 2019-12-02:
Hello,

i have a problem when i want to load a previously copied map via the python api.
For example, if I copy Town7 in the editor (e.g. Town07_2) and then load it via command

```
world = client.load_world("Town07_2")
map = world.get_map()
```

I get the following error message

```
ERROR: unable to parse the XML string 
	done!
Traceback (most recent call last):
  File "C:/Users/xxx/python/test/CarlaConnectionTest.py", line 13, in <module>

    map = world.get_map()
RuntimeError: failed to generate map
```

Does anyone have any idea why?

`world = client.load_world("Town07")` works fine.

Thanks a lot!

Chris

lechrizzle-87, 2019-12-04:
Okay, the corresponding .xodr file was missing. Now it works!

closed 2019-12-04 by lechrizzle-87

adi-geek2, 2019-12-05:
@lechrizzle-87 You mean , it was missing on the client side ?
If so , which location in client , Did you add the .xodr file ?

tom-bu, 2020-11-27:
You have to go to carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/ and add the .xodr file. If the new map is based on a default map, just copy the .xodr file from the original map and rename the file to match your map