carla-simulator/carla #2260: Error when "make launch"

roserbatlle, 2019-12-03:
Hey! 

I'm trying to build CARLA 0.9.5 on windows 10 and I keep getting this error when executing _make launch_: 

`C:\carla>make launch`
`/c/carla/Util/BuildTools/Setup.bat: line 1: @echo: command not found`
`/c/carla/Util/BuildTools/Setup.bat: line 2: setlocal: command not found`
`/c/carla/Util/BuildTools/Setup.bat: line 4: rem: command not found`
`/c/carla/Util/BuildTools/Setup.bat: line 5: syntax error near unexpected token '('
/c/carla/Util/BuildTools/Setup.bat: line 5: 'rem rpclib, gtest and boost libraries for CARLA (carla.org).'
make: *** [setup] Error 2`

Actually, I get the same error for any type of make command (e.g. make rebuild or make clean). I did install all the requirements to build the environment. I'm using Visual Studio x64 Native Tools Command Prompt on VS 2017. 

Any tips on what I might be doing wrong? Do I miss any necessary add-on? 

Thanks for your help!

closed 2020-02-08 by stale[bot]

Wogwan, 2020-08-11:
> Hey!
>
> I'm trying to build CARLA 0.9.5 on windows 10 and I keep getting this error when executing _make launch_:
>
> `C:\carla>make launch`
> `/c/carla/Util/BuildTools/Setup.bat: line 1: @echo: command [...]

Dear, did you fix it later? I met the same question in 2020/08/12. Feel free to let me know if any @ huanghejun097@gmail.com. Cheers

arielc-brillianetor, 2023-02-27:
Hi,
Don't know if I can still add a comment after the issue was closed but here is a solution. The same thing happened to me!
according to: "https://carla.readthedocs.io/en/0.9.0/how_to_build_on_windows/" instead of running:
`make PythonAPI`
you should directly run the setup.bat file which is in the `Util/BuildTools` folder. This file has some errors in it as well. First of all you need to redefine: `set ROOT_PATH=C:\carla` at the top of the file.
Besides that, all the places that call `%ROOT_PATH:\%` you should remove the `:\` so you stay only with the variable itself: `%ROOT_PATH%`.
This did the trick for me! 
Good luck