carla-simulator/carla #2269: Run carla faster for RL training

kargarisaac, 2019-12-04:
Hi,

I want to train an RL agent in Carla. Is there any way to run the simulation faster? 

I use:
carla9.6 
ubuntu 18.04
rtx 2080ti

Thank you

SajjadBaloch, 2019-12-05:
can you send your email i will tell solution their

kargarisaac, 2019-12-05:
@SajjadBaloch 
My email: kargarisaac@gmail.com
Thank you

ndeshp2s, 2019-12-05:
Hello @kargarisaac @SajjadBaloch. 

Even I am looking for a solution to train my RL agent faster in Carla. Could you please also share the solution with me? @SajjadBaloch my mail id: niranjan.deshpande187@gmail.com

Thank you in advance.

kargarisaac, 2019-12-09:
> can you send your email i will tell solution their

Hey. Did you get my email?

LuserBen, 2019-12-11:
Hello, @kargarisaac 
Could you please also share the solution with me? My e-mail id: Lbpeng@mail.dlut.edu.cn

Thanks very much!

hsezhiyan, 2019-12-17:
Hello @SajjadBaloch ,

Can you share the solution with me too: hsezhiyan@ucdavis.edu

Thank you!

weizhaoji, 2019-12-26:
Hello @SajjadBaloch ,

Can you share the solution with me too: 419365004@qq.com

Thank you!

Usaywook, 2020-02-18:
Hello @SajjadBaloch

ZhangYouRong, 2020-03-02:
Hello @SajjadBaloch ,
I have encountered the same problem, could you please share the solution with me ?
My email: zyr8@foxmail.com

laofuren, 2020-03-03:
Hello @SajjadBaloch ,

Can you share the solution with me too: 992100567@qq.com

Thank you!

closed 2020-03-03 by kargarisaac

Mulberry26, 2023-08-29:
> can you send your email i will tell solution their
@SajjadBaloch @ndeshp2s @kargarisaac @LuserBen @weizhaoji 
Could you guys  maybe share with me the RL accelerated time program? Thanks a lot!
my email is [armpro24@gmail.com
](url)