carla-simulator/carla #2278: how to add Town07 to 0.9.6 （carla 0.9.6 is an uncompiled version)

huangxueqin123, 2019-12-06:
I've already downloaded carla 0.9.6.tar.gz and Town07_0.9.6.tar.gz .When I decompress both of these, i tried to load town07 but failed. It says there is no town 07.

cassfalg, 2019-12-11:
There is an import script somewhere I think. Pretty sure it's mentioned in the docs for the additional towns. You don't need to decompress the town yourself, the script does that.

huangxueqin123, 2019-12-11:
Excuse me, do you know which script is or how to operate it?

huangxueqin123, 2019-12-11:
> There is an import script somewhere I think. Pretty sure it's mentioned in the docs for the additional towns. You don't need to decompress the town yourself, the script does that.

Thank you, however，do you know which script software is or how to operate it?

cassfalg, 2019-12-11:
https://carla.readthedocs.io/en/latest/asset_packages_for_dist/

This describes a lot of unrelated stuf, but also what you need to do I think. I've used from source versions for a long time, so I don't know what you need to do for the precompiled version.

huangxueqin123, 2019-12-11:
> https://carla.readthedocs.io/en/latest/asset_packages_for_dist/
> 
> This describes a lot of unrelated stuf, but also what you need to do I think. I've used from source versions for a long time, so I don't know what you need to do for the precompiled version.
thanks

closed 2020-02-16 by stale[bot]