carla-simulator/carla #2283: no member named 'PostPhysicsComponentTick'

jonlakun, 2019-12-10:
The following error appears when I make lauch, hope someone can help me.

tecnalia@U110593:~/carla$ make launch
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
Setup.sh: rpclib-v2.2.1_c2-c7 already installed.
Setup.sh: gtest-1.8.1-c7 already installed.
Setup.sh: recast-c7 already installed.
Setup.sh: CARLA version 0.9.6-53-g3e15ed3a.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/tecnalia/Escritorio/UnrealEngine/UnrealEngine-release'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/tecnalia/carla/Unreal/CarlaUE4'
bash "/home/tecnalia/Escritorio/UnrealEngine/UnrealEngine-release/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/tecnalia/Escritorio/UnrealEngine/UnrealEngine-release/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/tecnalia/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/Escritorio/UnrealEngine/UnrealEngine-release/Engine ~/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/tecnalia/Escritorio/UnrealEngine/UnrealEngine-release).
Performing 4 actions (8 in parallel)
[1/4] Compile Module.Carla.cpp
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:2:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.cpp:8:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.h:14:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/GnssSensor.h:9:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/Sensor.h:9:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.h:11:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/WorldObserver.h:9:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/DataStream.h:9:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/AsyncDataStream.h:13:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/streaming/Stream.h:9:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/streaming/detail/MultiStreamState.h:9:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/streaming/detail/StreamStateBase.h:10:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/streaming/detail/Session.h:9:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/streaming/detail/tcp/ServerSession.h:18:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/ip/tcp.hpp:25:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/ip/basic_resolver.hpp:28:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/ip/basic_resolver_query.hpp:21:
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/ip/resolver_query_base.hpp:19:
/home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/ip/resolver_base.hpp:69:5: warning: declaration shadows a variable in namespace 'boost::asio::ip' [-Wshadow]
    v4_mapped = BOOST_ASIO_OS_DEF(AI_V4MAPPED),
    ^
/home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/ip/address_v6.hpp:291:20: note: previous declaration is here
enum v4_mapped_t { v4_mapped };
                   ^
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:52:
/home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/PixelReader.cpp:57:3: warning: 'FTextureRHIParamRef' is deprecated: FTextureRHIParamRef typedef is deprecated; please use FRHITexture* directly instead. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile. [-Wdeprecated-declarations]
  FTextureRHIParamRef Texture = RenderResource->GetRenderTargetTexture();
  ^
/home/tecnalia/Escritorio/UnrealEngine/UnrealEngine-release/Engine/Source/Runtime/RHI/Public/RHIResources.h:1165:1: note: 'FTextureRHIParamRef' has been explicitly marked deprecated here
UE_DEPRECATED(4.23, "FTextureRHIParamRef typedef is deprecated; please use FRHITexture* directly instead.")
^
/home/tecnalia/Escritorio/UnrealEngine/UnrealEngine-release/Engine/Source/Runtime/Core/Public/Misc/CoreMiscDefines.h:227:43: note: expanded from macro 'UE_DEPRECATED'
#define UE_DEPRECATED(Version, Message) [[deprecated(Message " Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.")]]
                                          ^
In file included from /home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:57:
/home/tecnalia/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/Sensor.cpp:20:9: error: no member named 'PostPhysicsComponentTick' in 'UStaticMeshComponent'
  Mesh->PostPhysicsComponentTick.bCanEverTick = false;
  ~~~~  ^
2 warnings and 1 error generated.
Makefile:210: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/tecnalia/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

LawranceXcl, 2019-12-11:
same issue I have!

nsubiron, 2019-12-11:
You're running a newer version of Unreal Engine, Carla requires 4.22 at the moment.

jonlakun, 2019-12-13:
Thank you nsubiron
I don't know if it was an Unreal Engine version problem. But the thing is that I downloaded Unreal directly from their webpage, and even its name was unreal engine 4.22 could have something different.
To sum up, I restart all the installing process from the Carla's documentation, Build in Linux, and it works.
Thank you again.
Jon.

closed 2019-12-13 by jonlakun

qhaas, 2019-12-16:
Based upon a brief examination of the CARLA 0.9.6 codebase, that once deprecated, now removed API call isn't used elsewhere in the code (unless I'm missing something).  The attached patch file simply comments out the problematic portion of the code and allows compilation to proceed with UE 4.23.  Based on the other deprecation warnings generated, 4.24 will likely be more problematic.  Obviously, proceed with caution since upstream only supports UE 4.22 with CARLA 0.9.6.

[ue4233_PostPhysicsComponentTick_fix.txt](https://github.com/carla-simulator/carla/files/3965947/ue4233_PostPhysicsComponentTick_fix.txt)