carla-simulator/carla #2287: depth camera - distance to camera or to near clipping plane?

cassfalg, 2019-12-10:
I am stitching together multiple images to form a wide angle fisheye image. This is the rgb fisheye:
[image]
It uses equidistant projection with a fov of 192°. I gather pixels from a front, top, right, bottom and left image with 90° fov each to create those images. It's a process similar to cube mapping. In carla, I place 5 camera sensors which are rotated accordingly to produce cube images that are later combined to a fisheye image. Here is a visualisation of the source images - different colour means different source image. To get a feel for the process, imagine you're the fisheye camera and you're standing in a perfect cube such that your eyes are in the center of the cube. You're facing one side of the cube, which shows the front image (white). The cube to the right shows the right image (darkest grey) and so on... Here's the picture:
[image]
You can also see the borders in the rgb image due to the way reflection works. See https://github.com/carla-simulator/carla/issues/2185 for some things that improve rgb image quality related to post processing effects like vignetting.

Now i've taken a deeper look into the depth images, and you can see the borders between the different pictures in the depth images also. I do not understand why that is the case. I would expect the depth to be calculated from the 3D point in the scene to the position of the camera. In that case, there should be no visible border where I am stitching images.
[image]
For reference, to create this image I first calculate the distance in meters using the formula from the depth sensor documentation. Then I set all depth values greater than 200 meters to 200 meters. Then I apply the natural logarithm to all distance values and save the image using the viridis colourmap. Here is the stitched image in the original raw format that carla uses: 
[image]

Is it possible that the depth sensor shows the distance to the near clipping plane and not the distance to the camera? Because that would explain the images I am getting.

cassfalg, 2019-12-13:
I have tried to correct the depth distortions I am seeing by adding the distance from the camera to the near clipping plane to the depth of each cube image. Using trial and error it looks like the near clipping plane is somewhere around 0.5 meters from the camera? That sounds wrong (setting is 10 unity units, which should be 0.1 meters?), but looks more or less correct. Can anyone please elaborate on the correct near clipping plane distance for the depth sensor and/or the strange behaviour I am getting? The following image is with my correction code for 0.5 meters: 
[image]

closed 2020-02-18 by stale[bot]

cassfalg, 2020-02-20:
Ah, yes. I figured this out, forgot to mention it here: Turns out my first assumption was wrong. The depth in the depth or z buffer, and also aparently the depth reported by carla is always measured along the optical axis, so it is only correct for the center pixel. If you're familiar with projections, the depth reported for each pixel is the orthogonal projection of the correct depth onto the optical axis. All other pixels must be scaled accordingly or their depth is wrong.

Let fov be the field of view of the camera, (i, j) be the position of a pixel in the image, height be the resolution of the square image and depth be the distance in meters as calculated by the formula on the documentation, then:
We calculate the center of the image as center = height/2 - 0.5
The focal length (distance to image plane in pixels) is: f = tan(fov/2) * heigh/2
The diagonal distance in pixels by pythagoras is: diagDist = sqrt((i-center)² + (j-center)² + f²)
The true depth of the pixel then is: depth * (diagDist / f)

So to sum this issue up: The depth reported by the depth sensor is, at least according to the documentation: "The "Depth" camera provides a view over the scene codifying the distance of each pixel to the camera [...]". That is wrong. Either the documentation needs to clearly state what the output actually is, or the depth sensor needs to be changed to produce the correct output.

FrancoisMasson1990, 2020-02-25:
Hello @cassfalg, why do you define focal length like that ?
It should be f = height/(2*tan(fov)) right ?

cassfalg, 2020-02-25:
Uh... 
tan(fov/2) = (height/2) / f <=> f = (height/2) / tan(fov/2)
Looks like you're right! I probably didn't notice since my fov is 90° so tan(fov/2)=1... Thank you for pointing this out!

[depth_fix_definitions.pdf](https://github.com/carla-simulator/carla/files/4252429/depth_fix_definitions.pdf)
This PDF diagram might also help to understand the issue...

Shashikant255, 2020-03-27:
Is there any direct way to get fisheye image in carla 0.9.8?

PTorr, 2020-07-07:
@cassfalg really cool work.
Can you share the code?

pangxiao2013, 2020-07-19:
> Ah, yes. I figured this out, forgot to mention it here: Turns out my first assumption was wrong. The depth in the depth or z buffer, and also aparently the depth reported by carla is always measured along the optical axis, so it is only correct for the center pixel. If you're familiar with projections, the depth reported for each pixel is the orthogonal projection of the correct depth onto the optical axis. All other pixels must be scaled accordingly or their depth is wrong.
> 
> Let fov be the field of view of the camera, (i, j) be the position of a pixel in the image, height be the resolution of the square image and depth be the distance in meters as calculated by the formula on the documentation, then:
> We calculate the center of the image as center = height/2 - 0.5
> The focal length (distance to image plane in pixels) is: f = (heigh/2)/tan(fov/2)
> The diagonal distance in pixels by pythagoras is: diagDist = sqrt((i-center)² + (j-center)² + f²)
> The true depth of the pixel then is: depth * (diagDist / f)
> 
> So to sum this issue up: The depth reported by the depth sensor is, at least according to the documentation: "The "Depth" camera provides a view over the scene codifying the distance of each pixel to the camera [...]". That is wrong. Either the documentation needs to clearly state what the output actually is, or the depth sensor needs to be changed to produce the correct output.

what if  the image is not a square image?

cassfalg, 2020-09-17:
Well, most of those formulas work wether or not the image is square or not. You need to use the bigger of the horizontal and vertical fov for the calculation of f. Or just "embed" or enlarge your non-square image into a square one. If you understand what is done here it should not be hard to know what do do if you're image is not square.

goksanisil23, 2021-04-21:
Hi @cassfalg 

I have stumbled upon your post while searching for the true meaning of the "depth" value that is being reported via the depth sensor. Based on my research as well as your description, I understand that the encoded depth value coming from the z-buffer/depth sensor is the perpendicular distance to the near clipping plane, for that pixel (orthogonal prijection on to the optical axis)

With the carla depth camera, I'm basically trying to emulate a laser range finder (lidar), which reports the radial "true" distance for an object, from the center of the sensor. For this, I'm first applying an inverse projection via the camera intrinsic matrix (K) 

`K = [[f, 0, Cu],
     [0, f, Cv],
     [0, 0, 1 ]]`
`P = ( inv(K) * p2d ) * depth` 

Only doing this, gives me a "pointcloud" as below (top view and side view respectively). The camera is located at the origin of the scene, and 2 cubes are placed at the same x distance and slightly different y distance from the camera (cubes are of same height)  

[image]
[image]

After applying the depth correction factor you've mentioned above, I'm getting more "radial" pattern, however I see that the cube's point cloud is skewed now:

[image]
[image]

It would be great if you have any intuition on this.

Thanks and regards
Göksan

cassfalg, 2021-04-21:
I have never used a matrix notation for this... And it's been a while... I don't quite get what you're images are supposed to show... Regardless, some hints as far as I remember:

I calculated a roll and pitch angle for polar coordinates and used those to calculate the directional vector for the ray that created a pixel. Check out those two diagrams for definition of terms. Image calibration tells you how to calculate the roll angle via simple trigonometry (atan2(udist, vdist) function I think?). For the pitch angle you need a mapping between the radial distance on the image and the angle to the optical axis, I call this theta. This depends on your camera model, for equidistant (fisheye) projection r = theta, for pinhole r = tan(theta). With some scaling factor f somewhere (depends on resolution and field of view, r is in pixels in my case).

[image_calibration_definitions.pdf](https://github.com/carla-simulator/carla/files/6352855/image_calibration_definitions.pdf)

[camera_coordinates_definitions.pdf](https://github.com/carla-simulator/carla/files/6352850/camera_coordinates_definitions.pdf)

That camera intrinsic matrix  (and it's inverse) can not work by itself, because matrices represent linear functions, and for pinhole you need tan / arctan, depending on which direction you want. A graphics pipeline does this in the step where they modify the geometry so that it's in the unit cube viewing frustum thing, I forgot the details. I think Cem Yuksel mentioned this somewhere in his GPU Pipeline Explanation on his YouTube Channel, or you google how a GPU pipeline works elsewhere if you want details how those intrinsic matrices relate to my diagrams. There may be a better (matrix / vectorized) way to do it than my pixel by pixel polar coordinate code, but I never followed up on that.

goksanisil23, 2021-04-22:
Thanks a lot for the diagrams and the insight. I'm using the matrix notation for speed, and hoping to elevate these calculations to a shader level later on when the geometry is right. But if getting the correct radial distance requires non-linear calculations, I will change the approach.

cassfalg, 2021-04-22:
To clarify, that part is for getting a directional vector for a pixel's origin -- for generating a point cloud for example. It's not necessary just to calculate the correct depth data. That's what this comment is for: https://github.com/carla-simulator/carla/issues/2287#issuecomment-588784029

If you do this often you may want to cache the factor per pixel since it does not change, and calculating it requires sqrt, which may be expensive. Classic compute memory trade off. Also, I have not checked if carla has changed what their depth sensors report since the end of 2019 or so. In my opinion they should, or at least provide their own conversion function and clarify the documentation.

goksanisil23, 2021-04-23:
I dont think it has changed, Carla is just encoding what Scene Depth expression returns in Unreal. 

https://docs.unrealengine.com/en-US/RenderingAndGraphics/Materials/ExpressionReference/Depth/index.html

And yes, `(diagDist / f)` correction that is applied for each pixel better be cached.

vinodrajendran001, 2022-06-24:
> depth distortions I am seeing by adding the distance from the camera to the near clipping plane to the depth of each cube image. Using trial and error it looks like the near clipping plane is [...]

Hi, if in case the image is not fisheye and the H and W are different then how to calculate the true depth of the pixel?

tngh5004, 2023-08-10:
Hello, @cassfalg!
How can i make wide angle fisheye rgb camera in CARLA?
Or is it post-processing the output? 
Feel free to point me to a site or documentation for reference. 
Thanks for your help :)