carla-simulator/carla #2296: Question about networking for multiagent

a-jahani, 2019-12-11:
Does anyone know if Carla uses the built-in networking of unreal engine?

To my understanding, there is one simulator running and if there are multiple clients connecting to the simulator(server) the simulator sends the sensor data to each client. 
Isn't it better for multiple agents we use the unreal engine built-in networking?

Please fix me if I am wrong
Thank you

debnera, 2019-12-12:
CARLA does not use the built-in networking of Unreal Engine. 

The PythonAPI and Unreal networking are built for completely different purposes. The clients are simply Python scripts capable of receiving sensor data and controlling one or more actors of the simulator.

If CARLA used the built-in networking, each Unreal Engine client instance of the simulation would also be simulating physics and rendering graphics. The Python clients are not doing any of these calculations. In other words, even if you use the Unreal networking, you still need the Python clients to control the actors.

However, you are correct that the Unreal networking could be beneficial in multi-agent scenarios. In my master's thesis I integrated the built-in Unreal networking to CARLA and was able to significantly improve the performance by spreading out the GPU load of RGB and LiDAR sensors over multiple computers. However, this brings up issues with synchronization. My implementation is also already out-dated as the CARLA PythonAPI is changing so fast.

a-jahani, 2019-12-12:
@debnera Thank you for your comprehensive answer. I understand now.

The reason I brought that up is exactly the same thing you mentioned, spreading the load. I was working at a company using UE4 and we optimized for 90FPS for multiple clients in VR applications whereas here I was surprised adding couple agents will lower the FPS significantly. Unreal engine networking is highly optimized and fast. I agree synchronization would be an issue but there are many ways to make things synchronized in UE4!

Maybe the best approach is to have both approaches in Carla. We can run two/multiple simulator (connected through ue4 built-in networking) each has multiple clients connected to it (using Carla networking). The user decides how many clients connect to the simulator based on their setup (maybe they have multiple machines with GPUs and they want to make use of it). I think this might be the right approach to spread out the load and significantly improve the speed of Carla in the future.

Will this approach something that Carla developers are interested in?

Thank you again.
Ali

closed 2020-02-17 by stale[bot]

Croquembouche, 2024-10-17:
> CARLA does not use the built-in networking of Unreal Engine.
>
> The PythonAPI and Unreal networking are built for completely different purposes. The clients are simply Python scripts capable of [...]

Hi,

Is it possible for you to share this code?

Thank you