carla-simulator/carla #2300: Changes not getting reflected even after building successfully

YDSh510, 2019-12-15:
Hi,

I have done some changes in LibCarla and PythonAPI module and it is getting build successfully but I am unable to see the changes getting reflected. I added few debug msgs.

make LibCarla
make PythonAPI
make rebuild

Then run - 
python manual-control.py and CarlaUE4.uproject ( to start server)
But I am unable to see the debug logs in  manual-control.py terminal and server (message logs).

Please help.

nsubiron, 2019-12-15:
If you used `log_debug(...)` that only prints in debug builds (on unit tests), by default not in the release builds. You can change it here https://github.com/carla-simulator/carla/blob/b46117685e2e037dd6c0e6246998e5988bdc2bfb/LibCarla/source/carla/Logging.h#L18-L24

YDSh510, 2019-12-16:
@nsubiron  - Thanks for the above pointer. 

Just wanted to give brief background - 

1. I am trying to do some change w.r.t to weatherParameter (in Carla9.5),  
Changing 'HardRainSunset' to HeavyRainSunset' for which I changed in below three files  

../PythonAPI/crla/source/Weather.cpp
../LibCarla/source/carla/rpc/WeatherParameters.h
../LibCarla/source/carla/rpc/WeatherParameters.cpp

and then build 
make LibCarla
make PythonAPI
make rebuild

and then tried running -

python manual-control.py 
and CarlaUE4.uproject ( to start server)
But I am unable to see the new name getting reflecting in pygame. Am I missing anything here?

2. To debug the issue I was adding debug messages. Just wanted to check if  we can use log_debug(...) for files under PythonAPI folder also or is there any other way ?

Thanks.

nsubiron, 2019-12-17:
In principle changing those 3 files should be enough, in fact just changing Weather.cpp in PythonAPI folder should be enough to see the changes reflected. Difficult to say, could be you have the carla Python module installed? so you're still seeing an older version of the module instead of the one you just built.

Then, just for logging, use `log_warning` instead, that'll print even when called from Python.

closed 2020-02-22 by stale[bot]