carla-simulator/carla #2311: blueprint 'sensor.other.rss' not found

kingwingshome, 2019-12-21:
File "manual_control_rss.py", line 733, 
bp = world.get_blueprint_library().find('sensor.other.rss')
IndexError: blueprint 'sensor.other.rss' not found

    Platform(s): Ubuntu,
    Python version(s): 3.5
    Unreal Engine version: 4.22

jackbart94, 2020-01-07:
Hi,

Have you done
```sh
make LibCarla.client.rss
```

and then
```sh
make PythonAPI.rss
```

?

kbu9299, 2020-01-08:
I got the same problem. 

@jackbart94 
I downloaded the binary (CARALA_0.9.7).  At which folder should I run `make`?

jackbart94, 2020-01-08:
@kbu9299 all ```make``` have to be executed from the folder where you cloned/download the CARLA project.

sheepwxy, 2020-01-11:
> I got the same problem.
> 
> @jackbart94
> I downloaded the binary (CARALA_0.9.7). At which folder should I run `make`?

hello,have you solved this problem?

kbu9299, 2020-01-11:
@sheepwxy 
Yes, I could manage to run `make` from where I cloned Carla as follows:
```
make LibCarla
make PythonAPI
```

sheepwxy, 2020-01-11:
@kbu9299  Thanks for your reply!  It means.. I should use the source version but no complied version?

kbu9299, 2020-01-11:
@sheepwxy Yes, you need to clone Carla source (not compiled version). For the full compilation, you may also need to download Unreal Engine. In my case, I only tested the compilation for LibCarla and PythonAPI.

sheepwxy, 2020-01-11:
@jackbart94   thanks! I will have a try.

sheepwxy, 2020-01-13:
@kbu9299  It does help， thanks !

Martin20, 2020-01-22:
I got the same error. I did all your suggestions but still received the following error:

File "manual_control_rss.py", line 731, in __init__
    bp = world.get_blueprint_library().find('sensor.other.rss')
IndexError: blueprint 'sensor.other.rss' not found
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 92 (sensor.other.collision) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 94 (sensor.other.gnss)

Starfunk, 2020-02-12:
I am also receiving this error.

Bapt-GR, 2020-02-17:
Hi,
I also still have this problem, even though both `make [...].rss` were said to have been successful

gitmaster3000, 2020-04-01:
Same problem here. Can someone please look into this

closed 2020-06-07 by stale[bot]