carla-simulator/carla #2318: Static Blueprint has no label in Semantic Segmentation Map

doiken23, 2019-12-28:
Hi

I'm trying to increase the variety of objects in the world of Carla Simulator. To this end, I tried to spawn many static objects from blueprints library.
Finally, I implement the script like this:
```
# -------------------
# Spawn Other Objects
# -------------------
# take all the random locations to spawn
spawn_points = []
for i in range(args.number_of_others):
    if (i + len(self.others_list)) >= args.number_of_others:
        break 

    spawn_point = carla.Transform()
    loc = world.get_random_location_from_navigation()
    if (loc is not None):
        # loc.z += 0.5 
        spawn_point.location = loc
        spawn_points.append(spawn_point)

batch = []
for spawn_point in spawn_points:
    other_bp = random.choice(blueprints_others)
    batch.append(SpawnActor(other_bp, spawn_point))

results = client.apply_batch_sync(batch, True)
for i in range(len(results)):
    if not results[i].error:
        others_list.append(results[i].actor_id)
```
To implement it, I used https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/spawn_npc.py as a reference.

However, in the semantic segmentation map given by `save_to_disk` method, I found all of the labels on static objects are 0. I think it should be 3 (other class).

Do you have any idea to put the correct labels on the static objects?

Best regard,

FrancoisMasson1990, 2020-02-27:
@doiken23 Did you solve your issue ?

doiken23, 2020-02-28:
I have solved this problem by using the 3D bounding box information of static objects. Thank you for helping!

FrancoisMasson1990, 2020-02-28:
@doiken23  What did you modify in the carla code for the static objects because I got the following :
'Actor' object has no attribute 'bounding_box'

doiken23, 2020-02-29:
I check the bounding box of each object from the UE4 editor😉
Since the location of spawn points can be taken from python API, I can get the 3D bboxes.

closed 2020-05-06 by stale[bot]