carla-simulator/carla #2322: Simulator _crashes_ (just close the window after around 50 seconds)

michelpf, 2019-12-31:
I'm running the simulator in Surface Pro 7, with 8 GB and GPU Intel Iris 1 GB.
The simulator runs OK in manual mode, but even in manual ou in the simulation, after some seconds, around 50 seconds, just closes.

In the log, I couldn't find anything relevant.

Despite the recommendation of GPUs with more power, I read some comments that should work in my machinie (Cousera docs).

By the way, I tried lower the resolution, FPS and window size. No success at all. :(

Follow the log:

Log file open, 12/30/19 21:32:16
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VSPerf140.dll' (GetLastError=126)
LogWindows: File 'VSPerf140.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(0) SRandInit(0).
LogTaskGraph: Started task graph with 5 named threads and 14 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.195727
LogD3D11RHI: Aftermath initialized
LogPluginManager: Mounting plugin Carla
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin PerformanceMonitor
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogInit: Using libcurl 7.47.1
LogInit:  - built for x86_64-pc-win32
LogInit:  - supports SSL with OpenSSL/1.0.2g
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_IDN
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = false  - Libcurl will NOT verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = nullptr  - Libcurl will use whatever was configured at build time.
LogInit: Build: ++UE4+Release-4.18-CL-3832480
LogInit: Engine Version: 4.18.3-3832480+++UE4+Release-4.18
LogInit: Compatible Engine Version: 4.18.0-3709383+++UE4+Release-4.18
LogInit: Net CL: 3709383
LogDevObjectVersion: Number of dev versions registered: 16
[... 719 lines cut ...]
[2019.12.31-00.33.18:037][  6]r.EarlyZPassMovable = "1"
[2019.12.31-00.33.18:052][  6]foliage.DitheredLOD = "0"
[2019.12.31-00.33.18:053][  6]sg.PostProcessQuality = "0"
[2019.12.31-00.33.18:053][  6]sg.ShadowQuality = "0"
[2019.12.31-00.33.18:053][  6]sg.TextureQuality = "0"
[2019.12.31-00.33.18:053][  6]sg.EffectsQuality = "0"
[2019.12.31-00.33.18:053][  6]sg.FoliageQuality = "0"
[2019.12.31-00.33.18:054][  6]foliage.DensityScale = "0"
[2019.12.31-00.33.18:054][  6]grass.densityScale = "0"
[2019.12.31-00.33.18:054][  6]r.TranslucentLightingVolume = "0"
[2019.12.31-00.33.18:054][  6]r.LightShaftDownSampleFactor = "4"
[2019.12.31-00.33.18:055][  6]LogCarla: Error: Please regenerate the road in edit mode for 'CarlaMapGenerator2' actor
[2019.12.31-00.33.18:055][  6]LogCarla: Generated DCEL with: { 4 vertices, 8 half-edges, 2 faces }
[2019.12.31-00.33.18:060][  6]LogWorld: Bringing up level for play took: 5.427768
[2019.12.31-00.33.18:061][  6]LogCarlaServer: Sending 1 available start positions
[2019.12.31-00.33.18:061][  6]LogCarlaServer: Sending 0 sensor descriptions
[2019.12.31-00.33.18:069][  6]LogCarlaServer: Episode start received: { StartIndex = 1 }
[2019.12.31-00.33.18:069][  6]LogCarlaServer: Warning: Client requested an invalid player start, using default one instead.
[2019.12.31-00.33.18:071][  6]LogCarla: Changing weather settings to "ClearNoon"
[2019.12.31-00.33.18:071][  6]PIE: Warning: Mobility of /Game/Maps/RaceTrack.RaceTrack:PersistentLevel.LightSource : LightComponent0 has to be 'Movable' if you'd like to move. 
[2019.12.31-00.33.18:071][  6]LogCarlaServer: Ready to play, notifying client
[2019.12.31-00.33.18:074][  6]LogCarla: Found 1 PlayerStart positions for spawning vehicles
[2019.12.31-00.33.18:074][  6]LogCarla: Warning: We don't have enough spawn points (PlayerStart) for vehicles!
[2019.12.31-00.33.18:074][  6]LogCarla: To cover the 9 vehicles to spawn after beginplay, it will spawn one new vehicle each 3.000000 seconds
[2019.12.31-00.33.18:074][  6]LogLoad: Took 6.068222 seconds to LoadMap(/Game/Maps/RaceTrack)

michelpf, 2020-02-29:
I'm still with this issue. :(

closed 2020-05-06 by stale[bot]

kvlsky, 2023-07-06:
same. Windows 11, GeForce RTX 3060, 16GB RAM

kvlsky, 2023-07-18:
> same. Windows 11, GeForce RTX 3060, 16GB RAM

Fixed it by using UE 4.27 and adding env var 'UE4_ROOT', that directs to the UE installation folder

fjzs, 2023-07-18:
@kvlsky could you show what's your UE installation folder full path please?

tungyen, 2023-07-19:
@kvlsky Do you set the variable as a user variable or a system variable? And I also want to see the UE4 installation path, please.

kvlsky, 2023-07-19:
@fjzs @tungyen set it as described here: [image] [image] unfortunately I cannot find the exact link to the documentation, only have a screenshot

tungyen, 2023-07-20:
@kvlsky Does the user path you mark in gray also necessary? I do not use miniconda to run my program, python 3.6 directly instead. Thank you.