carla-simulator/carla #2324: many function mismatch between agent ,local_planner and misc

CliffBao, 2020-01-02:
In carla 0.9.7 I find that there is a function call due to misc change.
1. In carla/PythonAPI/carla/agents/tools/misc.py, function is_within_distance_ahead now get 3 inputs only, while in carla/PythonAPI/carla/agents/navigation/agent.py function _is_light_red_europe_style and _is_vehicle_hazard set 4 parameters, where the third one meaning orientation should be deleted and the first two parameters should be transform rather location type, which means get_transform() should be used to replace get_location()
You can see in the screenshot taken in master branch git.
[image]
[image]

2.  In carla/PythonAPI/carla/agents/navigation/local_planner.py  function set_global_plan, an error might occurs in a situation where user move one agent to another place and reset its destination. In  such a situation, set_destination will be called in basic_agent and set_global_plan function would be call in local_planner. However, although self._waypoint_queue is cleared, if self._waypoint_buffer is not empty, the next target waypoint would be old waypoint and the vehicle would move wrongly. So in this function, self._waypoint_buffer.clear() is also required.

3. In local_planner the target speed is set to 20km/h, I'm wondering whether it can be changed larger into 40, 60, or even 100 or not? In my experiment, I tried to tune it to 40, but the vehicle performs badly and keeps swinging left and right. Then I change it into default 20, and set basid_agent's target_speed into 40, the result was the same. So I guess this pid parameters are only for 20km/h target speed? And any target speed change in basic_agent is also permitted?

jackbart94, 2020-01-07:
Hi,

1. Yes, you are correct. It will be fixed in the upcoming days.

2. Not so sure about this one, I don't get the premise: why should the user move the agent to another place? A custom function would probably suffice.

3. Target speed can be changed, however yes you need to modify the PID values if you change it. Those are good only for 20km/h.

If you're interested, you can check [https://github.com/carla-simulator/carla/pull/2091](this agent) which has some more functions added.

closed 2020-03-14 by stale[bot]

qhaas, 2020-09-10:
Did the update that fixes this (0.9.7.3) ever find its way into the CARLA 0.9.7 docker container?  We are seeing this problem with the container, but not the 0.9.7.4 binaries.