carla-simulator/carla #2325: saving image with Bounding_Box Carla 0.9.6

stevelaclasse, 2020-01-03:
Hi, i am using the client_bounding_box.py to take image from the simulator. i have add the line to save the images taken by the camera, but it also wanted that these images should contain the Bounding_box of the vehicles and pedestrians, like there are displayed in the pygame interface.
Please how could i achieved this.

thanks

stevelaclasse, 2020-01-03:
i have replaced the pygame drawing functions with the carla debug.draw_box, and the box are displayed on th
[image]
e pictures taken by the camera.
but the other problem, is that these Boxes are to big, and i relly just want the 2D Boxes. 
if someone could take a look at it, i relly appreciate
[image]

n-akram, 2020-01-10:
A quick, but dirty solution is using pygame and pillow. You get the image from the active display and save it.

`import pygame`
`from PIL import Image`

`screen = pygame.display.get_surface()`
`data = pygame.image.tostring(screen, 'RGB')`
`w, h = screen.get_size() `
`img = Image.frombytes('RGB', (w,h), data)`
`img.save('sample.jpeg')`

This saves the image as shown. Make sure to hide HUD and all display messages.

[image]

stevelaclasse, 2020-01-12:
Thank you for your response. i have try it, but i always have the 3D Boxes, and there are other images without boxes. in 100 Images captured, 40 doesn't have boxes.
[image]

stevelaclasse, 2020-01-12:
Did you change the pygame **draw_bounding_boxes**  function ??

stevelaclasse, 2020-01-12:
i have modified the pygame draw_bounding_boxes function and now i am obtaining this. 
[image]

i will managed later to get ride of the boxes behind buildings.

n-akram, 2020-01-13:
Hi, 

Sorry for the delayed response. I was travelling. 

Yes we changed the client_bounding_box.py function (we are doing this in 0.9.7). 

For boxes of objects hidden in scene, we are planning to use semantic_segmentation along with bounding rectangles to filter out those that are not in scene. We have had no luck yet. 

Please let us know if you managed to get it.

stevelaclasse, 2020-01-13:
i have try this to get rid of object too far for the scene
i have changed this 
bounding_boxes = [bb for bb in bounding_boxes if all(bb[:, 2] > 0)]

by this : bounding_boxes = [bb for bb in bounding_boxes if all((bb[:, 2] > 0) and( bb[:, 2] < 60 ))]

i have also define the minimum area of  boxes for walkers and vehicles, because when there are too far from the scene thier area is small.  you have to test by your side to find the appropriate value.

let me know if you find a better solution.

stevelaclasse, 2020-02-06:
For those, who will be interested, this is how i got the 2D coordinates.

`    @staticmethod
    def get_bounding_2d_boxes_coordinates(bounding_boxes, type_object):
        """
                Draws bounding boxes on pygame display.
                """
        coordinates=[]

        for bbox in bounding_boxes:
            points = [(int(bbox[i, 0]), int(bbox[i, 1])) for i in range(8)]

            #print("points shape:", len(points))
            #print("type points:", type(points))
            #print("one point type:", type(points[0]))
            #print(points)
            min_x = points[0][0]
            min_y = points[0][1]
            max_x = points[0][0]
            max_y = points[0][1]
            for point in points:
                if (min_x >= 0):
                    if (point[0] >= 0 and point[0] <= min_x):
                        min_x = point[0]
                else:
                    if (point[0] >= 0):
                        min_x = point[0]

                if (point[0] >= 0 and point[0] >= max_x):
                    max_x = point[0]

                if (point[1] <= min_y):
                    min_y = point[1]
                if (point[1] >= max_y):
                    max_y = point[1]
            if (min_x < 0 or min_x == max_x):
                print()
                #print("unable to find extrema values:", min_x, max_x, min_y, max_y)
            else:
                print("rectangle points:", min_x, max_x, min_y, max_y)
                point1 = (min_x, min_y)
                point2 = (min_x, max_y)
                point3 = (max_x, max_y)
                point4 = (max_x, min_y)

                if(type_object==1):

                    if((max_x-min_x)>=12) and ((max_y-min_y)>=20):

                        coordinates.append(point1+point3)
                    else:
                        print("car coordinates too far from the scene:",min_x, max_x, min_y, max_y)
                else:
                    if ((max_x - min_x) >= 12) and ((max_y - min_y) >= 12):

                        coordinates.append(point1 + point3)
                    else:
                        print("walker coordinates too far from the scene:", min_x, max_x, min_y, max_y)

                #pygame.draw.line(bb_surface, BB_COLOR, point1, point2)
                #pygame.draw.line(bb_surface, BB_COLOR, point2, point3)
                #pygame.draw.line(bb_surface, BB_COLOR, point3, point4)
                #pygame.draw.line(bb_surface, BB_COLOR, point4, point1)
        return coordinates
`

Harsharma2308, 2020-03-17:
what is type object and how do you get that value?

closed 2020-05-30 by stale[bot]