carla-simulator/carla #2326: FPS calculation for server and client

johnzjq, 2020-01-06:
Hi Pros,
We are trying to increase the frame rates of the server and multiple clients. However, based on some intensive experiments, we cannot fully figure out what is going on behind these numbers. Would you please provide me with some clue?  Thanks in advance.
The server runs on a windows pc using the unreal editor (4.9) and all clients (0.9.6) are either running on the same pc (LOCAL) or on a linux pc (REMOTE). The following are the statistics. To maximize the server performance, we packaged the server code into an EXE. What is obvious at this moment is the network overhead is the bottle net in the REMOTE rgb mode.

REMOTE LiDAR mode
| client number | server fps | client fps | server CPU % | serve GPU % | Network % |
|---------------|------------|------------|--------------|-------------|-----------|
| 0             | 80         |            |              |             |           |
| 1             | 77         | 28         | 38           | 33          | 1         |
| 2             | 77         | 30         | 38           | 34          | 2         |
| 3             | 77         | 21         | 38           | 34          | 4         |
| 4             | 77         | 11         | 35           | 34          | 4.5       |
| 5             | 77         | 7          | 41           | 36          | 5         |

LOCAL LiDAR mode
| client number | server fps | client fps | server CPU % | serve GPU % |
|---------------|------------|------------|--------------|-------------|
| 0             | 105        |            |              |             |
| 1             | 94         | 17         | 38           | 58          |
| 2             | 90         | 17         | 52           | 55          |
| 3             | 80         | 14         | 67           | 51          |
| 4             | 65         | 13         | 80           | 42          |
| 5             | 13         | 57         | 72           | 39          |

REMOTE RGB mode
| client number | server fps | client fps | server CPU % | serve GPU % | network % |
|---------------|------------|------------|--------------|-------------|-----------|
| 0             | 80         |            |              |             |           |
| 1             | 50         | 35         | 44           | 68          | 95        |
| 2             | 35         | 35         | 44           | 68          | 98        |
| 3             | 27         | 35         | 44           | 68          | 98        |
| 4             | 23         | 35         | 44           | 68          | 98        |
| 5             | 20         | 35         | 44           | 68          | 98        |

LOCAL RGB mode
| client number | server fps | client fps | server CPU % | serve GPU % |
|---------------|------------|------------|--------------|-------------|
| 0             | 105        |            |              |             |
| 1             | 37         | 62         | 36           | 48          |
| 2             | 20         | 62         | 48           | 48          |
| 3             | 14         | 62         | 56           | 50          |
| 4             | 9          | 62         | 64           | 52          |
| 5             | 7          | 26         | 73           | 53          |

johnzjq, 2020-01-06:
it seems the markdown tables are not parsed.

debnera, 2020-01-07:
Not sure I can help that much, but this is a topic I have been meaning to investigate further.

LIDAR:

Not much to say here. LiDAR uses cpu only. Your client seems to be the bottleneck in both local and remote scenarios. Lower quality settings, simpler simulatio scene and no-rendering mode might increase server fps.

RGB:

Its clear that the performance drops rapidly when number of cameras is increased.

I may be wrong, but I believe this has something to do with the drawcall overhead of rendering from multiple cameras and/or translating all objects to multiple camera views. In this case it is not specifically a CARLA issue, but can most likely be replicated in any Unreal Engine project (or any other game engine). Using less vertices and objects in scene might reduce the amount of drawcalls and therefore improve fps.

I have only tested this on Linux and OpenGL, but the fps drops very quickly when you increase the amount of cameras, even with very low resolution (e.g. 100x100). Higher resolution causes higher GPU load.

You could try using the Unreal Engine profiler to see where most of the time is spent. https://docs.unrealengine.com/en-US/Engine/Performance/Overview/index.html

closed 2020-03-14 by stale[bot]