carla-simulator/carla #2328: Incorrect doppler velocities

shubham294, 2020-01-07:
Hi everyone,

The current radar implementation seems to calculate the range rate along a given azimuth angle as simply the velocity the actor instead of velocity *along* the line of sight with radar. Does anyone know if fixing this is in the development pipeline of Carla?

[image]
X-axis: azimuth in degrees, Y-axis: range rate. One plot each for the front and the rear radar respectively. I was expecting a sinusoid relation b/w range rate and the azimuth angle.

I'd be happy to fix this if it's not planned.

jackbart94, 2020-01-07:
Hello,

I think @marcgpuig can be more helpful than me with this issue, however you can always try to fix it and open a pull request, which we will later review and eventually update the code with.

germanros1987, 2020-01-08:
@marcgpuig please, let's work on fixing this issue.

doterop, 2020-01-10:
I am looking into this. I hope I can update you on Monday.

doterop, 2020-01-10:
@stackSmasher294 Can you send me the script that you use to test? Or at least the part that you use to visualize the radar output?

stackSmasher294, 2020-01-18:
Um, the actual script just reads off a ROS topic. It was used in conjunction with the Carla ros-bridge.

stackSmasher294, 2020-01-18:
I think I may have figured out where the problem is. Just confirming - to test it, I'll have to build CARLA from source, right?

doterop, 2020-01-20:
Yep, because the logic of the radar is in the server side.

stackSmasher294, 2020-01-20:
Rebuild Carla with the fix and the problem is solved! The doppler plots now look like this:

[image]

I'll open a pull request with the fix.

doterop, 2020-02-03:
Thanks @stackSmasher294 ! I just pushed it to master.

closed 2020-04-11 by stale[bot]