carla-simulator/carla #2335: Low quality settings persist after switching back to Epic

BenAscent, 2020-01-07:
After running Carla in low quality using the following command, any subsequent run of the simulator retains the low quality level even if Epic quality is specified.

```
./CarlaUE4.sh -quality-level=Low
```

It looks to happen because the `~/.config/Epic/CarlaUE4/Saved/Config/LinuxNoEditor/GameUserSettings.ini` file is saved with the following content:

```
[ScalabilityGroups]
sg.ResolutionQuality=100.000000
sg.ViewDistanceQuality=3
sg.AntiAliasingQuality=3
sg.ShadowQuality=0
sg.PostProcessQuality=0
sg.TextureQuality=0
sg.EffectsQuality=0
sg.FoliageQuality=0
```

But when Carla is started again in Epic quality, those settings are not replaced by their initial values (quality of 3 for all sg), making the low quality mode persisting after the first use, unless the ini file mentioned above is manually cleared.

My suspicion is that the settings from this file are applied after the code that handles `quality-level`, but I haven't tried to verify that.

This occurs with version 0.9.7 on Linux (Ubuntu 18.04)

germanros1987, 2020-01-08:
We are working on fixing this very annoying issue.

BenAscent, 2020-01-09:
@germanros1987 Thank you!

Note that it seems that calling
```
Scalability::ToggleTemporaryQualityLevels(true);
```
Before applying quality level settings seem to solve the problem. I tried to put it in `ApplyQualityLevelPostRestart()` just before the switch `statement`.

This function seems to prevent saving the settings change into the user file. You'll need to include `Scalability.h`

closed 2020-03-09 by bernatx