carla-simulator/carla #2337: Couldn't change the town in CARLA 0.9.7

vivekjaiswal003, 2020-01-07:
Hi,

I recently started working with Carla0.9.7
In previous versions I would change the town by the following command

./CarlaUE4.sh  /Game/Carla/Maps/Town01  -benchmark -fps=20 -quality-level=Epic

But in 0.9.7 version this is not working. Every time Town03 is loading irrespective of the town code mentioned in the command. Thanks in advance.

jackbart94, 2020-01-07:
Hi,

To change the map, you can use the config.py script located inside PythonAPI/utils. This will allow you to change several things at run time, including the map. Let me know if that works for you!

vivekjaiswal003, 2020-01-11:
@jackbart94 Thanks for your feedback.
Actually after going through previous issues. It came to my knowledge that command `./CarlaUE4.sh /Game/Carla/Maps/Town01 -benchmark -fps=20 -quality-level=Epic` is no longer supported. Now map  could only be changed through PYTHONAPIs by 
`world = client.load_world('Town02')`

closed 2020-01-11 by vivekjaiswal003