carla-simulator/carla #2338: Can I spawn traffic lights or traffic signs with Python API?

doiken23, 2020-01-07:
Hi. Thank you for providing a great simulator. We are enjoying using it!

Now we're trying to change the number of objects in the simulator world. For example, we can change the number of vehicles or walkers by spawning or destroying them with Python API as implemented in [this sample code](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/spawn_npc.py).

Here, can we spawn traffic lights or traffic signs in the same way as vehicles or walkers? 
If we can, could you show us some examples?
If we can't, could you tell us if there are any alternative solutions?

Best regards,
Thank you.

jeffreykxiao, 2020-01-10:
Forgive this fairly unresearched answer, but considering that spawnActor only allows you to spawn actors inside of the blueprint library and that the [blueprint library](https://carla.readthedocs.io/en/latest/bp_library/) doesn't contain traffic lights / signs, I suspect that you cannot do such a thing.

I see that there are a few actors inside of the static category, such as traffic cone and traffic warning, but it is likely that these are just static meshes and would have no impact on any in-Carla vehicle AI.

At this time, I can only suggest that you should build an editor version and create a new map with the signs you need - or extended the blueprint library to allow users to spawn the correct types.

doiken23, 2020-01-10:
ありがとうございます😉

Finally, I edit my map manually. That is enough at this point. Later I will work on extending the blueprint library.

closed 2020-01-10 by doiken23

aravindSwamy94, 2020-05-27:
Hi @doiken23 : Great to hear that you have added it manually..
Can you please explain it how??

doiken23, 2020-05-28:
I opened UE4 editor with `make launch` and put objects manually.

aravindSwamy94, 2020-05-28:
Hi.. Thank you for your response... It would be great if you share some link on this as in the steps to do those manual editing..

doiken23, 2020-05-29:
Maybe this video is useful for you (https://www.youtube.com/watch?v=EMkfCCFMtPI)

fezsid, 2023-03-29:
Can you please explain how you are exporting the map after editing it?