carla-simulator/carla #2339: CARLA 0.9.7 does not generate a python library on Windows 10

gokhanbarisaker, 2020-01-07:
Hi,

Running `make PythonAPI` claims to complete successfully with following output.

```
-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:/path/to/Workspace/Cpp/carla/PythonAPI\carladist"! 
```

At first sight, the mixed usage of path separators looks suspicious. However, even if I ignore the forward slashes (i.e., /) by replacing them with backwards slash (i.e.,\), there is no `carladist` in that path.

As user, I would expect there is an actual output in the highlighted path given with the build output.

# Test environment

Windows 10 1909
Visual Studio 2017
Unreal Engine 4.22.3
Python 3.7.4.final.0
GNU Make 4.2.1
cmake version 3.12.18081601-MSVC_2
carla 0.9.7 (b9fbbf7fd03ee2a4f3a7baf4343a28381cfe81ef)

CodingdAwn, 2020-01-08:
same problem and i have some link error
but finally output successful
-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:/project/carla/PythonAPI\carladist"
checking and it's nothing generated

lease\source/libcarla\libcarla.cp38-win32.exp
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp_PyBool_FromLong
libboost_python38-vc141-mt-x64-1_69.lib(class.obj) : error LNK2001: 无法解析的外部符号 __imp_PyBool_FromLong
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp_PySlice_Type
libboost_python38-vc141-mt-x64-1_69.lib(function_doc_signature.obj) : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libboost_python38-vc141-mt-x64-1_69.lib(stl_iterator.obj) : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libboost_python38-vc141-mt-x64-1_69.lib(enum.obj) : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libboost_python38-vc141-mt-x64-1_69.lib(pickle_support.obj) : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libboost_python38-vc141-mt-x64-1_69.lib(str.obj) : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libboost_python38-vc141-mt-x64-1_69.lib(builtin_converters.obj) : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libboost_python38-vc141-mt-x64-1_69.lib(list.obj) : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libboost_python38-vc141-mt-x64-1_69.lib(function.obj) : error LNK2001: 无法解析的外部符号 __imp_PyErr_Occurred
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp_PyEval_SaveThread
libboost_python38-vc141-mt-x64-1_69.lib(pickle_support.obj) : error LNK2001: 无法解析的外部符号 __imp_PyExc_RuntimeError
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp_PyExc_RuntimeError
libboost_python38-vc141-mt-x64-1_69.lib(errors.obj) : error LNK2001: 无法解析的外部符号 __imp_PyExc_RuntimeError
libboost_python38-vc141-mt-x64-1_69.lib(class.obj) : error LNK2001: 无法解析的外部符号 __imp_PyExc_RuntimeError
libboost_python38-vc141-mt-x64-1_69.lib(function.obj) : error LNK2001: 无法解析的外部符号 __imp_PyExc_RuntimeError
libcarla.obj : error LNK2001: 无法解析的外部符号 __imp_PyLong_FromLong
libboost_python38-vc141-mt-x64-1_69.lib(enum.obj) : error LNK2001: 无法解析的外部符号 __imp_PyLong_FromLong
libboost_python38-vc141-mt-x64-1_69.lib(function_doc_signature.obj) : error LNK2001: 无法解析的外部符号 __imp_PyLong_FromLong

alexmargulis, 2020-01-08:
I think that this is a known issue, therefore there are no compiled versions greater than 0.9.5. for Windows.
I'm personally stuck with these errors:

   Creating library build\temp.win-amd64-3.6\Release\source/libcarla\libcarla.cp36-win_amd64.lib and object build\temp.win-amd64-3.6\Release\source/libcarla\libcarla.cp36-win_amd64.exp
libpng.lib(pngwrite.obj) : error LNK2001: unresolved external symbol deflate
libpng.lib(pngwutil.obj) : error LNK2001: unresolved external symbol deflate
libpng.lib(pngwrite.obj) : error LNK2001: unresolved external symbol deflateEnd
libpng.lib(png.obj) : error LNK2001: unresolved external symbol inflateReset
libpng.lib(pngrutil.obj) : error LNK2001: unresolved external symbol inflateReset
libpng.lib(png.obj) : error LNK2001: unresolved external symbol crc32
libpng.lib(pngread.obj) : error LNK2001: unresolved external symbol inflate
libpng.lib(pngrutil.obj) : error LNK2001: unresolved external symbol inflate
libpng.lib(pngread.obj) : error LNK2001: unresolved external symbol inflateEnd
libpng.lib(pngread.obj) : error LNK2001: unresolved external symbol inflateInit_
libpng.lib(pngwutil.obj) : error LNK2001: unresolved external symbol deflateReset
libpng.lib(pngwutil.obj) : error LNK2001: unresolved external symbol deflateInit2_
build\lib.win-amd64-3.6\carla\libcarla.cp36-win_amd64.pyd : fatal error LNK1120: 9 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\BuildTools\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:/carla/PythonAPI\carladist"!

alexmargulis, 2020-01-09:
There is a workaround for this problem - run PythonAPI 0.9.7 compiled for linux in WSL

jeffreykxiao, 2020-01-10:
The message is definitely confusing, because the .egg file is saved to `carla/PythonAPI/carla/dist`. It's a fairly innocuous error in the build python API batch script. Thanks for posting this, I also had issues with PythonAPI building on windows (though I was able to resolve my problems). I will be submitting a PR soon to resolve some problems I discovered while looking into this issue.

jeffreykxiao, 2020-01-10:
I can't say for sure that #2348 will fix all of the issues mentioned by users here but I hope it will help. I encountered many issues when I upgraded an earlier version where I had an installed PythonAPI. In the past I had used `make clean` but I didn't know that it wasn't working. With the change, `make clean` should remove any intermediate files messing with your build. `make PythonAPI` has always worked fine for me. I am using Win10 Pro x64, UE4.22.3, VS2017 w/ Native Tools x64 Prompt, python 3.6, Make for Windows 3.81

arguskos, 2020-01-16:
@alexmargulis  Can you please explain how to run it. I mean I can compile it under wsl but it will not run client because there is no support for graphics in wsl as I understand and you can't launch it from windows as well?

alexmargulis, 2020-01-21:
@arguskos 
https://github.com/microsoft/WSL/issues/2855#issuecomment-358861903

wiescho, 2020-01-23:
@CodingdAwn I managed to build the PythonAPI by deleting the headers option in the boost build script Util\InstallersWin\install_boost.bat
in my case with this option boost won't build the libboost_python lib

echo %FILE_N% Building...
b2 -j%NUMBER_OF_ASYNC_JOBS%^
    --layout=versioned^
    --build-dir=.\build^
    --with-system^
    --with-filesystem^
    --with-python^
    --with-date_time^
    architecture=x86^
    address-model=64^
    toolset=%TOOLSET%^
    variant=release^
    link=static^
    runtime-link=shared^
    threading=multi^
    --prefix="%BOOST_INSTALL_DIR%"^
    --libdir="%BOOST_LIB_DIR%"^
    --includedir="%BOOST_INSTALL_DIR%"^
    install

quote from http://boost.2283326.n4.nabble.com/doc-b2-headers-td4702968.html
> `b2 headers` is not really intended for end users.
It's only relevant when working from git as the
packaged releases have all the headers moved into
boost/ to begin with.  
>

PR #2348 (2020-02-03): Fix BuildPythonAPI cleaning and success message

#### Description

BuildPythonAPI.bat was broken because it was expecting `%PYTHON_LIB_PATH%` to have a trailing slash, which it was missing. 

Fixes #2339  

Behavior prior to this change:

- `make clean` does not clean the intermediate files for building PythonAPI
- `make PythonAPI` displays the wrong path for the build result

Behavior with this change:

- `make clean` properly cleans up the intermediate files located in `PythonAPI/carla/build`, `PythonAPI/carla/dist`, and `PythonAPI/carla/source/carla.egg-info`
- `make PythonAPI` reports the  [... cut]
closed 2020-02-03 by bernatx