carla-simulator/carla #2341: [carla0.9.7] Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API

sheepwxy, 2020-01-08:
hello,  I installed the new version 0.9.7 besides 0.9.6.   However, when i run the script in the PythonAPI. I got the error as below, anyone can help me?

> WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.6 
WARNING: Simulator API version  = 0.9.7 
Segmentation fault (core dumped)

jeffreykxiao, 2020-01-10:
When you say you have both versions "beside" each other, do you mean you have both egg files for the PythonAPI? If your `PythonAPI/carla/dist` folder contains multiple .egg files of different versions, it may cause unintended behavior.

https://github.com/carla-simulator/carla/blob/b9fbbf7fd03ee2a4f3a7baf4343a28381cfe81ef/PythonAPI/examples/spawn_npc.py#L17

This line shows where the script searches for the egg, note the *-matching, which would pick up both your 0.9.6 and 0.9.7 eggs. The code then takes the first index of whatever `glob` gets you, which I assume could be either of the two eggs, my guess would be lexographical order so 0.9.6. Either way, I think you should first check what eggs you have in the dist folder (it may just be the case that 0.9.7 did not build properly and you don't have the right egg) and then perhaps attempt hard-coding the path to the 0.9.7 egg into the script for testing purposes.

sheepwxy, 2020-01-11:
> When you say you have both versions "beside" each other, do you mean you have both egg files for the PythonAPI? If your `PythonAPI/carla/dist` folder contains multiple .egg files of different [...]

Thanks for your help. And your way does help.

closed 2020-01-11 by sheepwxy

QianwenZhu, 2021-06-28:
make sure you run with the same carla version and python version.

ChristofKaufmann, 2022-01-05:
After an update of carla, you have to update the python package manually, like:
```bash
cd /opt/carla-simulator/PythonAPI/carla/dist
pip install carla-0.9.*-cp37-cp37m-manylinux_2_27_x86_64.whl
```

ronyshaji, 2022-12-01:
@jeffreykxiao Did you able to find the solution for this ? I am still looking for an answer.  in My case it showing like this:
`WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 0.9.12
WARNING: Simulator API version  = 7b5bfac-dirty`

ronyshaji, 2022-12-01:
> After an update of carla, you have to update the python package manually, like:
> 
> ```shell
> cd /opt/carla-simulator/PythonAPI/carla/dist
> pip install carla-0.9.*-cp37-cp37m-manylinux_2_27_x86_64.whl
> ```

Do you have the code for windows build ?

DeLeonOscar, 2023-06-27:
@ronyshaji did you solve the problem?

DeLeonOscar, 2023-06-27:
> @jeffreykxiao Did you able to find the solution for this ? I am still looking for an answer. in My case it showing like this:
>
> ```
> WARNING: Version mismatch detected: You are trying to connect to a [...]

do you solve the problem? i am getting a similar error