carla-simulator/carla #2342: Semantic segmentation - Car windows

carlaUser, 2020-01-08:
I have noticed that the car windows are completely transparent to the segmentation camera, i.e. the segmentation algorithm will label the corresponding pixels as whatever is behind them, **regardless of how opaque the windows are**.

In the following pictures, we can see that, despite of the windows being completely visible and near opaque in the RGB image, part of the windshield is labeled as road (which, in turn, is barely visible).

[image]
[image]

Is this the intended behavior? Otherwise is any fix planned?

jeffreykxiao, 2020-01-09:
I can confirm this issue. In my opinion, it should not be the intended behavior as a car window should for all intents and purposes still be semantically part of a vehicle. 

The fix is simple:
[image]
Translucent materials have an additional property to allow custom depth writes (this incurs a performance cost as they are normally excluded from the custom depth pass since they need to be handled last). As long as the base material for all car glass, which I believe is Static/Vehicles/GeneralMaterials/Glass.uasset has this flag changed, all material instances using Glass.uasset will be rendered in custom depth.

However, as Carla content is distributed through AWS / Google Drive, there is no way for a user to push this fix. If you have an editor build you can change this for yourself, but otherwise you will need to wait on a new asset pack / new release.

meteorshowers, 2020-02-07:
could you please share your fixed version?
Thanks!

guygo12345, 2020-03-31:
Hey!
I'm also interested. If you can would love to get the fix :)

closed 2020-06-06 by stale[bot]

magehrig, 2020-06-18:
This issue still persists in 0.9.9

guygo12345, 2020-07-07:
It seems like this issue isn't going to be solved, but meanwhile I'm attaching here the updated uasset file.
If you want to use it just replace the original with this
[glass.zip](https://github.com/carla-simulator/carla/files/4883228/glass.zip)