carla-simulator/carla #2343: Parked cars are not actors

carlaUser, 2020-01-08:
The parked cars loaded by default in the maps are not listed as actors by `world.get_actors()`. This is due to the fact that it is only the static meshes that are deployed in the map in the Unreal Editor, instead of the corresponding actors (due to performance reasons, I guess?).

The problem with this is that they are not accessible from the Python client as far as I know, and I cannot, therefore, get their positions or bounding boxes when generating the ground truth of a scene.

Is there any way to access this information (without having to alter the maps in the editor) that I might have missed?

jeffreykxiao, 2020-01-09:
Unfortunately, I don't believe there is any way to handle this from outside the Unreal project. You're correct about the proximal reason: the parked cars in the maps are indeed StaticMeshActors and Carla's API abstracts away the Unreal notion of "actor" in favor of their own Actor/Vehicle/etc classes used in their EpisodeState and ActorList. When you query the Carla World class, it sends you info from these.

However, based on the documentation it seemed to me that the `get_actors` function you're calling *should* be able to get other UE4 actors in the scene. I was able to find the culprit in CarlaEpisode Line196:

https://github.com/carla-simulator/carla/blob/954f36b6a039134e15c2a08945ad8517a871f691/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.cpp#L196

This `InitializeAtBeginPlay()` function is responsible for registering other actor types in the ActorList and this line ensures that we don't get static meshes with mobility set to Static/Stationary (like buildings and the such). Unfortunately, the parked car StaticMeshActors also have mobility set to Static, so they get excluded (mistakenly, IMO) by this code. 

My suggestion would be for the parked cars to have their mobility changed to Movable, so that they will no longer be ignored. This is both simple and resolves your problem. I would also be fine with the concept of parked cars being removed entirely from maps and instead transferred to an external concept such as a Scenario and spawned in at the start of a run.

stevelaclasse, 2020-02-27:
I have edited the those cars with the mobility Movable, but they don't get them in the actor list, i d´think because they are tagged as "Static Mesh" in Unreal.

the other method would be to remove all the parked cars from the Maps, but would have preferred letting  there and get their  information.

jeffreykxiao, 2020-02-27:
Yeah, looking at the code it would seem that even if they are set as movable, they'll still be of type "prop" in the blueprint API (since all StaticMeshActors get tagged as props), meaning they won't show up as actors in the actor list. What a bummer.

stevelaclasse, 2020-02-27:
in which file could it be possible to edit those properties ?

stevelaclasse, 2020-03-09:
I have edited the Maps of Town 1 to 5 and removed all the parked cars,  

https://github.com/stevelaclasse/Carla_Test/blob/master/Edited%20Maps%200.9.6.zip

xinshuoweng, 2020-05-03:
having static vehicles parked on the map is good and increases the diversity and reality of the data. So it would be very nice if the Carla team can help add features that can control those vehicles in the future release.

germanros1987, 2020-06-26:
I agree. @XGodina @doterop let's make all our vehicles actors, even if by default they are in an "idle" state.

JMorceaux, 2020-06-26:
> I have edited the Maps of Town 1 to 5 and removed all the parked cars,
> 
> https://github.com/stevelaclasse/Carla_Test/blob/master/Edited%20Maps%200.9.6.zip

Thanks, really helpful !

xinshuoweng, 2020-08-30:
> I agree. @XGodina @doterop let's make all our vehicles actors, even if by default they are in an "idle" state.

Thanks, German! It would be nice to see this feature of controlling static vehicles as actors soon.

p-testolina, 2020-09-04:
I'm working on the bounding box extraction as well, this would be a nice feature to have!
Thanks for your work guys.

arashroomi, 2020-09-24:
I have a similar issue but I am interested to extract more than just parked cars. In specific buildings, parked vehicles, and other obstacles. Is there a solution to extract their bounding boxes before rendering?

stevelaclasse, 2020-09-28:
I had tried to edit parked cars properties from UE4 engine, i have tried to assign some properties of the mobile cars to the parked cars for them to be recognized by the Carla API,  but with no success.
I have only made some change in the properties, but with changes also in the categories and classes, it may be possible to achieved it.

doterop, 2020-10-06:
Hi to everyone,

We added a method in our last release 0.9.10 to retrieve all the bounding boxes of all the elements of the level. So, if you need to get the bb of the parked vehicles call:
`bboxes = world.get_level_bbs(carla.CityObjectLabel.Vehicles)`
This call will include every vehicle in the level, so if you only need the parked vehicles call it after loading the level and before spawning any other.

Besides, we are working on the possibility to disable specific elements of the level, thus you can disable specific parked cars or all of them, and convert the parked cars into actors so you can control them. But this will come on the next release 0.9.11.

Ping me if you need any help.

tinmodeHuang, 2020-10-07:
> Hi to everyone,
>
> We added a method in our last release 0.9.10 to retrieve all the bounding boxes of all the elements of the level. So, if you need to get the bb of the parked vehicles call:
> `bboxes [...]

could you tell me when it will be released?

arashroomi, 2020-10-07:
> Hi to everyone,
>
> We added a method in our last release 0.9.10 to retrieve all the bounding boxes of all the elements of the level. So, if you need to get the bb of the parked vehicles call:
> `bboxes [...]

So will it also contain the bounding boxes of the buildings and trees and other objects?

doterop, 2020-10-07:
Hi @tinmodeHuang ,
> could you tell me when it will be released? in addition, is that to say that I can easily have access to states of all actors such as posititon, velocity and heading, etc?

The 0.9.11 should be released between the end of this year and beginning of the next one. Unfortunately, I can't be more precise about it.
When you say actors here, I guess that you mean Unreal actors right? Because you already can know the position, orientation and velocity of a Carla actor. If this is not the case, can you elaborate the question a bit further please?

Hi @arashroomi ,
> So will it also contain the bounding boxes of the buildings and trees and other objects?

This is already added. There is a function called `get_level_bbs` in world and you can filter it by the object type. Is the same type used in the semantic segmentation.
Example:
```
// Retrieve the BBs of all the vehicles of the level. Spawned cars and parked cars
vehicles_bbs = world.get_level_bbs(carla.CityObjectLabel.Vehicles)

// Retrieve the BBs of all the traffic lights of the level (only the box that contains the lights).
tl_bbs = world.get_level_bbs(carla.CityObjectLabel.TrafficLight)
```
There is a new enum called `CityObjectLabel` where all the object types are defined.

tinmodeHuang, 2020-10-07:
@doterop they refer to actors in Carla such as vehicle, motorcyclist, pedestrian, bicyclist and even parked car. In my implementation of reinforment learning, it acquires real-time information about surrounding traffic aprticipants in Carla to help make a decision.

doterop, 2020-10-07:
@tinmodeHuang So, right now with the last release, you could only get the BBs of the parked vehicles and that will give you the position and orientation.

tinmodeHuang, 2020-10-07:
> @tinmodeHuang So, right now with the last release, you could only get the BBs of the parked vehicles and that will give you the position and orientation.

in that case, only velocity of parked vehicles remains to be determined, so I just call that function above after loading the level and before spawning any other during the beginning of simulation to get it

doterop, 2020-10-07:
Yes @tinmodeHuang ,
Parked vehicles will not move, so their velocity will be always zero.

tinmodeHuang, 2020-10-08:
it was appreciated. so if I only need real time states of those actors around the ego car rather than those of all actors in the map at every time step, can I get those by the call above or by just adjusting it a little bit?

tinmodeHuang, 2020-10-13:
it turns out that the bounding boxes of all of actors specified are returned only once instead of all the time at fixed frequency after calling get_level_bbs() once

doterop, 2020-10-19:
Can you show at example please to clarify your question?

stevelaclasse, 2021-01-12:
Thank you first of all for providing us the coordinate of parked cars, i have try et_level_bbs() for vehicles and somehow the client_bounding_box.py in Example Folder (that i have updated to work with the new function) does the wrong calculation and the bounding boxes are really bad located.
I have tested the two method:

- Only active Cars :  self.world.get_actors().filter('vehicle.*')                                                 --> Good BBox
- All cars (with parked cars) : self.world.get_level_bbs(carla.CityObjectLabel.Vehicles)      --> Bad BBox

germanros1987, 2021-01-26:
@doterop could you please follow up on this to close this issue for good?

doterop, 2021-01-27:
Hi @stevelaclasse 

Which values did you receive from both functions on the same bounding box?
Could you provide a bit more info with code or images?

Also, the bounding box that you get from the actor is in local coordinates while the bounding boxes from `get_level_bbs` are in world coordinates since they don't have any reference to any actor.

closed 2021-03-23 by doterop