carla-simulator/carla #2344: Mintlinux 19.3, Carla 0.9.7, Unrealengine 4.22,  'Make PythonAPI' failed

zarod, 2020-01-08:
- The default python3 linked to python 3.6.9; I changed the default python3 to 3.5.9 but failed as well.
- The default python linked to python 2.7.17
- I use clang-7 and clang++-7
'make launch' finished successfully
'make PythonAPI'  faild
I got the following error:

`BuildLibCarla.sh: Success!
BuildPythonAPI.sh: Building Python API for Python 2.
Traceback (most recent call last):
  File "setup.py", line 131, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 72, in get_libcarla_extensions
    raise NotImplementedError
NotImplementedError
Util/BuildTools/Linux.mk:80: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1`

Any idea appreciated!

robogast, 2020-01-09:
It fails on the check on line 34 in `setup.py`: `if linux_distro.lower() in ["ubuntu", "debian", "deepin"]`  
what is your OS?

zarod, 2020-01-10:
Thank you for the tip! 
My mintlinux use the kernel for ubuntu 18.04 but I will install the ubuntu and test it. which ubuntu version you recommend?

robogast, 2020-01-10:
I don't think its absolutely necessary to change OS's because of this simple check, I would first simply comment out the check, try recompiling and see if it works:
```python
# linux_distro = platform.dist()[0]
linux_distro = "ubuntu"
```
If that doesn't work, you can always still switch OS.

zarod, 2020-01-13:
Changed it but got some other errors. Ubuntu based system is not enough. must be ubuntu exactly. 
thnx for the tips mate!

dmatos2012, 2020-02-02:
@robogast, thank you for that tip, that was all that was needed. Thanks

robogast, 2020-02-14:
In lieu of an official solution, I made the following diff: [distro-check.txt](https://github.com/carla-simulator/carla/files/4204145/distro-check.txt)

So, the easiest way to fix this is: (with 'easiest' meaning: 'without having to edit `setup.py` yourself')
1. download the `distro-check.txt`, and place it in `carla` root
1. `pip install distro && pip2 install distro`
1. from `carla` root: `git apply distro-check.txt`
1. `make PythonAPI`

closed 2020-04-23 by stale[bot]

philippwulff, 2020-11-09:
I also had this issue. I am using Linux Mint 19.3 Cinnamon and needed to add "linux mint" to the list