carla-simulator/carla #2346: Adding additional Actors

tmofle, 2020-01-09:
Hello,

I am using Carla as a driving simulator for a distracted driving research project. I was curious if it is possible to add additional actors in the simulator (through my own code python code) , such as a drone. The only requirement for the drone is to fly across the roadway. Would this be possible using CARLA?

jeffreykxiao, 2020-01-09:
In order for you to use a proper static mesh of a drone, you'll need to import that into the Carla Unreal project itself and make sure it's in the Blueprint Library so you can spawn it from python. 

There's no method for controlling a flying vehicle so I think you would want to spawn a static mesh prop, and then use "set transform" -esque commands to move your drone around. In order to test this out, you could first try using a generic prop that's already in Carla's library (https://carla.readthedocs.io/en/latest/bp_library/#static) that hopefully resembles a drone at a distance, perhaps the Bike Helmet or something similar. Unfortunately Carla's spawning API does not allow scaling of the actor, so you won't be able to change the shape of the spawned object.

tmofle, 2020-01-12:
Thank you for your assistance. I think this will help my project greatly.

closed 2020-01-12 by tmofle