carla-simulator/carla #2351: Is it possible to steer car in CARLA 0.9.6  by providing input from CSV file which has steering angle, velocity, brake, gear and timestamp.

surendrasah, 2020-01-10:
Hi Team,
I am using CARLA simulator 0.9.6 in windows and I have a CSV file which has necessary parameter (velocity, steering angle, brake, gear and timestamps) required to steer the vehicle. Any help on this would be really appreciated

n-akram, 2020-01-13:
I am not sure if there is PythonAPI for windows. But if there is, process should be fairly simple.

`import csv`
`f = open('steering.csv', 'r')`
`self._control.steer = float(f.readline())`

surendrasah, 2020-01-24:
Thanks @n-akram. It will not work like this. I have found a way to do this. I need to iterate around the throttle, brake, gear and steering value.
But i still have few doubts.
I have four brake value from four tires. how do i convert it to a single value as CARLA expects value from [0,1].
I have steering angle but its in positive sign like 20, 30, 120 etc. degree. how would i calculate if its turn left or right. Any idea on this would be really appreciated.

closed 2020-03-18 by surendrasah

akramfaouji, 2021-03-31:
@surendrasah hey, i'm in the same state as you , i wish if you find a way to run carla with a csv file , if you have Any idea on this would be really appreciated.