carla-simulator/carla #2353: [0.9.7] TrafficManager doc

sheepwxy, 2020-01-11:
Hello, I am using the Compiled version 0.9.7 Carla. I want to use the new feature "Traffic Manager", but I can not get any doc or demo. Can anyone help me? @Praveen Kumar

jackbart94, 2020-01-13:
Hi,

Sorry about that, docs are coming quite soon. The steps however are quite the same as before, you just launch the spawn_npc.py and the Traffic Manager will be working. For more specific features, the docs will be updated!

sheepwxy, 2020-01-13:
@jackbart94   Thanks for your work.  I have another question, should I use the source version(no complied version) to use the traffic manager function?  In the "spawn_npc.py" script in complied version, I did not find API(such as force_lane_change) about the traffic manager.

jackbart94, 2020-01-13:
There are two different scripts for now, spawn_npc.py hides some of the features that TM has, but keeps the original structure of the code.

Meanwhile, in the tm_spawn_npc.py you have an instance of TM and using this you can apply parameters (such as force_lane_change) just by changing a few lines of the code. We ought to make this a bit easier for users to access in the upcoming future, and also provide tutorials!

talsperre, 2020-03-01:
@jackbart94  thanks for the work. Is there any tentative timeline by which you would be releasing the traffic manager documentation?

@sheepwxy, were you able to find the API in the `tm_spawn_npc.py` file?

closed 2020-05-07 by stale[bot]