carla-simulator/carla #2356: Semantic segmentation under bridge

wzhang1, 2020-01-13:
To reproduce, in a town with bridges, driver under a bridge, e.g. in town04,
we can see that the ceiling below the bridge is marked to class 6 with is a lane marker,
although the other side is a road(class 5), we may not prefer to define them as class 6.
[image]
[image]

magehrig, 2020-04-27:
Can confirm this issue under version 0.9.8

magehrig, 2020-05-10:
@marcgpuig Can this be reopened? It's not fixed afaik.

marcgpuig, 2020-07-06:
Sorry for the late response, I'll try to communicate it to the art team ASAP. They plan to iterate fixing several of these issues son.

DnDare, 2020-07-07:
Greetings,  we've made changes to SS of the bridge
[image]

It is merged into master , it will be released in future updates

DnDare, 2020-07-07:
The bridge SS, has been changed to WALL
[image]

magehrig, 2020-07-08:
Has this been done for all the bridges in all the towns or just this particular one?

magehrig, 2020-07-13:
For example this is in Town05
[image]

So it does not appear to be fixed. Can we reopen the issue since it is not solved?

DnDare, 2020-07-13:
I'll have a look at this tomorrow, I'll post some updates

DnDare, 2020-07-14:
[image]
The bridge has been changed,  to be reviewed and approved  yet ,( will be updated in  future releases)

magehrig, 2020-07-16:
Great, thanks for addressing this issue. Are these changes already available on the current master branch?

DnDare, 2020-07-16:
Yes, they're in art's master branch,  if you have access to private towns, there  are some redirectors that affects T09

magehrig, 2020-07-16:
@DnDare Which branch is "art's master" branch? I am interested in using Carla with the corrected segmentation maps. Can I do that? If yes, how?

DnDare, 2020-07-16:
We have 2 public master branches,  one is for code the other is for art assets, you just will have to update as usual.
If you have any problem, just ask

magehrig, 2020-08-12:
Just to inform you, I did what you recommended but in Town 05 there were still road lines under the bridge

DnDare, 2020-08-12:
Sorry to hear that, and for the inconveniences.
I'm not sure what to recommend you, these changes have been made, and a few others. By the end of July we made a pre-release, it is meant to have this changes, but there are some issues, and this is why it wasn't announced. (I think it's public)
Now the team is mostly on vacations, I guess the future release will have the changes, and I'm guessing it will be in September.
If you can wait I'd suggest to wait, else you can try your luck, sry about not being able to help more.

DnDare, 2020-09-28:
@magehrig  Greetings, I guess you already noticed, we've launched Carla 0.9.10. 
There're improvements to SS, hope they suit your needs:
http://carla.org/2020/09/25/release-0.9.10/

closed 2020-12-25 by stale[bot]