carla-simulator/carla #2358: How to increase pygame window

obi5656, 2020-01-15:
carla ver0.9.6

manual_control_steeringwheel.py is used.
I want to increase the number of pygame windows to 3 as I want to actually drive.
I want to use three monitors to see not only the front but also the left and right. Is there a way? Help me.

Sk8t0s, 2020-01-15:
you can play wit hthe window size but it will just extend 1 view not sure you can have multiple display maybe with more camera but someone more expert can confirm that but otherwise you can set de resolution when launching the pygame

n-akram, 2020-01-20:
Its not possible to have multiple pygame windows. Atleast not that I am aware of. What you can do instead is extend the size of one window to occupy three monitors and then blit three images from three camera views side by side.

closed 2020-03-27 by stale[bot]