carla-simulator/carla #2364: config.py has issues setting the framerate

exoticDFT, 2020-01-17:
**Version**: 0.9.7
**Platform**: Linux (Probably all platforms, but I cannot test)

There seems to be an issue with the way "framerate" is set with the `config.py` script found in the util folder of the PythonAPI. Specifically, when setting the value to 0, in order to regain the variable framerate of the server an error occurs. 

> Traceback (most recent call last):
  File "../util/config.py", line 251, in <module>
    main()
  File "../util/config.py", line 222, in main
    if settings.fixed_delta_seconds > 0.0:
TypeError: unorderable types: NoneType() > float()

This same error occurs when trying to use `-delta-seconds` or the "default" flag, `-d`. The only way to recover variable framerate seems to be restarting the server. It also seems that the `-benchmark -fps=xx` flag for `CarlaUE4.sh` does not set the framerate for the server. Maybe this was removed from previous versions? Is there a full changelog for releases to help with determining the breaking changes?

Lastly, this may need to be a separate issue, but when setting the framerate for the simulator, only certain dynamics seem to change appropriately. For example, it seems the water ripples, spectator camera controls and even vehicle controls are influenced by the framerate. However, the ephemeris model (or sun dynamics) is unchanged by the framerate/simulation time. I'm sure there are other dynamics that may not change appropriately, but I haven't had a chance to dig any further.

closed 2020-03-24 by stale[bot]

thillRobot, 2020-12-04:
I experiencing similar behavior in version 0.9.10.1. The command `python util/config.py --host 192.168.x.y --fps 20` runs without error and returns a message `set fixed frame rate 33.33 milliseconds (30 FPS)`, but there is no visible affect in the server or client view. Is this a known issue? There is a similar comment in #2509.

thillRobot, 2020-12-04:
Update: I just found #1021. This helps me understand.