carla-simulator/carla #2367: Mounting camera statically

Harsharma2308, 2020-01-19:
I need to mount camera on a pole at an intersection. (The reason is to get a pseudo data collection pipeline ready, and I could test my downstream algorithms like object detection and tracking.)
How can I mount a sensor to any static point in the scene ? (I know there are APIs available for mounting on a vehicle.)

n-akram, 2020-01-20:
You can find the transfornation of the pole and use that to statically place a camera at that position. There is some trial and error involved. But you can get it working. Following example works is used to set up a camera on traffic light.

```python
p0 = traffic_light.get_transform()
blueprint_library = self.__world.get_blueprint_library()
camera_bp = blueprint_library.find('sensor.camera.rgb')
camera = self.__world.spawn_actor(camera_bp, p0, attach_to=traffic_light)
```

You will have to play around with transformation.

Harsharma2308, 2020-01-25:
Thanks for this.

closed 2020-01-25 by Harsharma2308

thinker310, 2024-06-27:
> ```python
> p0 = traffic_light.get_transform()
> blueprint_library = self.__world.get_blueprint_library()
> camera_bp = blueprint_library.find('sensor.camera.rgb')
> camera = self.__world.spawn_actor(camera_bp, p0, attach_to=traffic_light)
> ```
```python
client = carla.Client('127.0.0.1', 2000)
client.set_timeout(20.0)
world = client.get_world()
traffic_light=world.get_actors().filter('traffic.traffic_light')[0]  #world.get_actors() will show all actors in the world, find the actor's type, then filter this type, the information will be shown.
p0 = traffic_light.get_transform()
blueprint_library = self.__world.get_blueprint_library()
camera_bp = blueprint_library.find('sensor.camera.rgb')
camera = self.__world.spawn_actor(camera_bp, p0, attach_to=traffic_light)
```
the completed code is above, thank you