carla-simulator/carla #2372: Autopilot mode working too slow in synchronous_mode example

aakashsehgal, 2020-01-20:
I am using Carla 0.9.6 in windows.

It provides an example of synchronous_mode.py. If i activate the autopilot mode and disable the waypoint updation, the whole simulation becomes extremely slow. 

```
vehicle.set_simulate_physics(False)
vehicle.set_autopilot(True)
```

Am i missing something ?

closed 2020-05-30 by stale[bot]

jht1759, 2020-08-21:
The same Issue is found when the synchronous mode is set. Is there any solution for it

MaXiaoSIOM, 2021-03-16:
The same issue with Version 0.9.11. I spawned 30 auto-piloted vehicles and 0 walkers with example script 'spawn_npc.py'  . In synchronous_mode,  almost 50ms taken for one simulation frame, while it is 6ms in no_synchronous_mode.