carla-simulator/carla #2379: make launch error

Martin20, 2020-01-22:
I went step by step in the tutorial to build Carla. I have done all steps as https://carla.readthedocs.io/en/latest/how_to_build_on_linux/ says. However, I got the error below:

[2020.01.22-02.52.23:551][  0]LogLinux: Warning: MessageBox: Vulkan Driver is required to run the engine.: Message:
[2020.01.22-02.52.23:551][  0]LogCore: FUnixPlatformMisc::RequestExit(bForce=true, ReturnCode=1)
[2020.01.22-02.52.23:551][  0]LogCore: FUnixPlatformMisc::RequestExit(1)
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 1

I tried all suggestions but did not work.

Martin20, 2020-02-03:
Thanks,

What is the right command?

jeffreykxiao, 2020-02-04:
@Martin20 https://carla.readthedocs.io/en/latest/configuring_the_simulation/#graphics-quality

opengl param mentioned here, just run it with carla.sh

closed 2020-04-11 by stale[bot]

KomalBagai, 2020-10-29:
@jeffreykxiao There is no such file as carla.sh. As said here, [](https://carla.readthedocs.io/en/latest/build_linux/) that `make launch `will open the editor, I tried:
`make launch -opengl4`
But I was getting the same error. Please help.

m4tice, 2020-11-25:
Hi, I'm building Carla on a remote PC (Ubuntu 18.04), following the latest Carla Docs and facing probably the same issue while running 'make launch'. I have tried erasing all and reinstalling but the error remains. The error is as below. 

...
[2020.11.25-14.35.48:370][  0]LogInit: Using OS detected language (de-DE).
[2020.11.25-14.35.48:370][  0]LogInit: Using OS detected locale (de-DE).
[2020.11.25-14.35.48:371][  0]LogTextLocalizationManager: No specific localization for 'de-DE' exists, so the 'de' localization will be used.
[2020.11.25-14.35.48:372][  0]LogTextLocalizationResource: LocRes '../../../Engine/Content/Localization/Editor/de/Editor.locres' could not be opened for reading!
[2020.11.25-14.35.48:372][  0]LogTextLocalizationResource: LocRes '../../../Engine/Content/Localization/EditorTutorials/de/EditorTutorials.locres' could not be opened for reading!
[2020.11.25-14.35.48:373][  0]LogTextLocalizationResource: LocRes '../../../Engine/Content/Localization/Keywords/de/Keywords.locres' could not be opened for reading!
[2020.11.25-14.35.48:373][  0]LogTextLocalizationResource: LocRes '../../../Engine/Content/Localization/Category/de/Category.locres' could not be opened for reading!
[2020.11.25-14.35.48:373][  0]LogTextLocalizationResource: LocRes '../../../Engine/Content/Localization/ToolTips/de/ToolTips.locres' could not be opened for reading!
[2020.11.25-14.35.48:373][  0]LogTextLocalizationResource: LocRes '../../../Engine/Content/Localization/PropertyNames/de/PropertyNames.locres' could not be opened for reading!
[2020.11.25-14.35.48:373][  0]LogTextLocalizationResource: LocRes '../../../Engine/Content/Localization/Engine/de/Engine.locres' could not be opened for reading!
[2020.11.25-14.35.48:398][  0]LogSlate: New Slate User Created.  User Index 0, Is Virtual User: 0
[2020.11.25-14.35.48:398][  0]LogSlate: Slate User Registered.  User Index 0, Is Virtual User: 0
[2020.11.25-14.35.49:081][  0]LogLinux: Warning: MessageBox: Vulkan Driver is required to run the engine.: Message: 
[2020.11.25-14.35.49:081][  0]LogCore: FUnixPlatformMisc::RequestExit(bForce=true, ReturnCode=1)
[2020.11.25-14.35.49:081][  0]LogCore: FUnixPlatformMisc::RequestExit(1)
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 1

Has anyone been able to solve it yet? Any help is appreciated. :D

HaoZhouGT, 2021-06-16:
Hey there, did you solve this? Same issue for me. Looks like we need to find the correct Vulkan driver for our own GPU. Is there a general guidance on this? Or we have to switch to opengl?

Kin-Zhang, 2022-01-29:
> Hey there, did you solve this? Same issue for me. Looks like we need to find the correct Vulkan driver for our own GPU. Is there a general guidance on this? Or we have to switch to opengl?

I also met this problem on one computer but other computer with same GPU didn't appear this error on `vulkan driver is required to run the engine`

Fatmangh, 2022-05-19:
Hi, I am running into this same issue [here](https://github.com/carla-simulator/carla/issues/2379#issue-553277201) I tried the suggested -opengl option but still see same. I am running the build version CARLA version 0.9.11-dirty with UnrealEngine_4.24. I followed [this tutorial](https://cfanatic.medium.com/building-carla-on-debian-gnu-linux-11-68d68fddf862) because I am running on a google cloud vm with Debian GNU/Linux 10. I also followed [this tutorial](https://rhyce.dev/2020/06/12/enable-opengl-and-disable-vulkan-in-unreal-engine-4-25/) to enable OpenGL on  UnrealEngine.