carla-simulator/carla #2381: Left hand drive in Unreal Open Drive Actor

gjossep, 2020-01-22:
Im having an issue with the Open Drive Actor in Unreal editor.

 I have created a road layout in RoadRunner and have exported to Unreal. The issue is that the road layout uses Left-Hand traffic and the automatic route generator seems to be ignoring the left hand drive of the roads. 

It seems to be a issue in the CARLA plugin, if i change the source of the plugin will it also compile if I compile CARLA from src?

jeffreykxiao, 2020-01-29:
1. Make edits to LibCarla to support LHT
2. `make clean`
3. `make LibCarla`
4. `make launch`
5. `make pythonAPI` 
6. run example

jeffreykxiao, 2020-01-29:
Carla doesn't support left-handed traffic at this time. Also Roadrunner exports the outdated 1.4 OpenDrive spec which is missing the road direction tag (Roadrunner uses the userdata header and a custom traveldir tag). However, you could force LibCarla to assume the input is LHT instead of RHT. The process you need to follow is:

1. Make edits to LibCarla to support LHT
2. `make clean`
3. `make LibCarla`
4. `make launch`
5. `make pythonAPI`
6. run example

gjossep, 2020-01-30:
What would the changes need to be in the LibCarla? Im unfamiliar with the structure of the LibCarla folders and how OpenDrive works.

jeffreykxiao, 2020-01-31:
I wasn't really familiar with it either but I managed to figure it out. I have a working version that I can put up in a few days. It's not something mergeable into carla proper because it's just a hack that forces LHT and breaks RHT

Abanoub-G, 2020-01-31:
That would be very useful. Or even just a brief explanation of the how the Carla Plugin interprets the OpenDrive file.

gjossep, 2020-02-04:
Do you have a working version yet that I could use? Or just a guide? I dont mind that it breaks RHT

jeffreykxiao, 2020-02-05:
@gjossep Thanks for the reminder, I'm working on a different project these days so kind of forgot about this. I made a Draft Pull Request with the process and change, in addition to suggestions for the Carla Team. Hopefully we can raise this issue instead of relying on my simple hack.

https://github.com/carla-simulator/carla/pull/2442

gjossep, 2020-02-12:
Hi,
I tried getting your pull to work, but the Open Drive Actor in unreal is still putting it in RHT instead of LHT. I followed all your instructions, etc. I cloned your copy of the carla and used that but still same results. Is there any major change I need to do to Unreal after doing the Make LibCarla?

closed 2020-04-19 by stale[bot]