carla-simulator/carla #2389: Windows client does not utilize command line arguments?

roschler, 2020-01-23:
I'm running Carla 0.9.5 on Windows 10/64-bit and I am trying to connect to a server instance.  I'm not using 0.9.7 because the person that is running Carla in server mode can only get 0.9.5 running on his AWS EC2 box.

It doesn't seem to me that the Windows version is using the command line arguments?  Here's why I say that.  (Note, obviously `{host address redacted}` had the real server IP address, which I am removing from view in this post):

I don't think Carla64.exe (Windows) is looking at the command line arguments anyways.  I tried "--host {host address redacted} --port 2000".  No luck.  I then tried Window style "host={host address redacted} port=2000".  No luck.  I looked at `manual_control.py` and it's looking for `host` and `port` in the `args` variable.  But I'm not convinced the Windows version, at least, is using them.  In all the trials I mentioned above, and a few others, the client took the same amount of time to pop up with the same empty world I see when I ran Carla without any command line arguments at all (i.e. - running in "local" mode).

I even tried some completely bogus command line arguments and bogus `host` and `port` values  too.  I never got any complaints or errors from Carla, nor did the client hang for a while like  you would expect it to when attempting to connect to a server at a bad host/port address.

Is the Windows EXE using the command line arguments in reality?

roschler, 2020-01-25:
Is anyone maintaining the Windows version?  Seems like a simple question I asked above.

bernatx, 2020-01-27:
Hi, we are working on keeping the windows version as updated as the ubuntu one, doing a lot of changes in the build pipeline to get that goal. 

In the meantime, I can tell you that there is no difference connecting a client from the windows or ubuntu versions. The client needs the host IP and port to connect to the server. 

The first thing that comes to my mind is that the security group of the EC2 instance where the server is running has not the port 2000 open to the world, preventing any client to connect. Can you check that?

Thanks,
Bernat

closed 2020-04-03 by stale[bot]