carla-simulator/carla #2396: Can't run CARLA 0.9.7 on Ubuntu 19.10 with RTX2070

kartikye, 2020-01-24:
Error occurs when i run `./CarlaUE4.sh`:

```
4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x2600041
  Serial number of failed request:  380
  Current serial number in output stream:  386
Signal 11 caught.
Segmentation fault (core dumped)
```

System is Ubuntu 19.19 with RTX 2070

daedric, 2020-01-27:
You should try to run it with  -opengl. carla uses vulkan by defualt, and I do have the same problem as well

jediofgever, 2020-03-18:
@how to run with opengl ?

closed 2020-05-24 by stale[bot]