carla-simulator/carla #2397: Using Town6 and Town7 in CARLA 0.9.7

ndeshp2s, 2020-01-24:
IS there a way to use towns 6 and 7 in Carla 0.9.7 version build from source? I can run these town in pre-build carla-0.9.7. But if I use its source version, these towns are missing.

krmartin, 2020-01-26:
Those towns don't come with the initial source code clone of CARLA 0.9.7 from GitHub.  You have several options to get them:

A. Run the ./Update.sh script in Linux as per instructions, to download the additional content *.tar.gz file and the script will automatically extract it into the correct location (and will delete the tar file afterwards).

Subsequent runs of ./Update.sh will only update as needed, by comparing a hidden file (./Unreal/CarlaUE4/Content/Carla/.version) of the content version you last Updated to the git tag version inside ./Util/ContentVersions.txt. 

B. Look inside ./Util/ContentVersions.txt and download (e.g. curl or wget) and extract the content for your particular version of CARLA manually. 

C. Click the link to the "AdditionalMaps_0.9.7.2.tar.gz" file here (or whatever the latest minor version is available for 0.9.7), and then extract the contents yourself:
https://github.com/carla-simulator/carla/releases/tag/0.9.7

D. If you're a developer, or feel comfortable with using/managing another lfs git repository, clone the towns and other assets here:
https://carla.readthedocs.io/en/latest/how_to_build_on_linux/#assets-repository-development-only

E. Carefully copy your existing copies of the Towns to the correct location from your binary download, assuming they're there. Doesn't seem like a good way to do it, unless you have a really slow Internet connection and just want to reuse what you already have.

ndeshp2s, 2020-01-27:
Thank you for your detailed explanation @krmartin

closed 2020-04-03 by stale[bot]