carla-simulator/carla #2401: How to do transform to geoposition

fyj97, 2020-01-27:
Hello CARLA team.
I see that there is an API can transform geoposition to transform.
But how can I do transform to geoposition.
Thank you very much

jackbart94, 2020-01-27:
Hi,

If I understand correctly your question, [the transform method is the answer](https://carla.readthedocs.io/en/latest/python_api/#carlatransform-class). Let me know if it helps!

fyj97, 2020-01-27:
Thanks a lot.
I am afraid I do not talk clearly. I see there is a function calls transform_to_geolocation Carla.map class, can converts a given carla.Location (x, y, z) to a carla.GeoLocation (lat, lon, alt). But I want to get the carla.Location (x, y, z) from  carla.GeoLocation (lat, lon, alt).

closed 2020-04-03 by stale[bot]

nuomizai, 2020-10-26:
> Hello CARLA team.
> I see that there is an API can do transform to geoposition.
> But how can I do geoposition to transform.
> Thank you very much

Hi, @fyj97 . Have you found ways to convert geoposotion to transform? I'm also troubled by this problem.