carla-simulator/carla #2403: Seg faults after python scripts

amstrudy, 2020-01-27:
I've recently begun having issues when running Python scripts with CARLA. I'll boot up CARLA (`make launch-only`), and then if I run one of the example scripts (`tm_spawn_npc.py` for example), it will run fine but once it finishes there is always an error. Sometimes its a simple `Segmentation fault (core dumped)` but sometimes its 
```Fatal Python error: PyEval_SaveThread: NULL tstate

Aborted (core dumped)
```
This happens with any script I try. All of the Python scripts definitely finish before the error occurs. I am running Ubuntu 18.04 and Python 3.6.9. Before posting this issue I pulled from the CARLA repo and ran ./Update.sh, and then rebuilt. To be clear - this isn't really affecting anything while I use CARLA, but as a result CARLA often freaks out that actors are missing.

jackbart94, 2020-01-27:
Hello!

We are aware of this problem, and are working towards fixing it. We know what is causing it, luckily as you said it doesn't affect the script's execution, but we are definitely going to patch this error!

amstrudy, 2020-01-28:
Thank you! Good to know it's not just me.

versenyi98, 2020-03-23:
Hi!
Is there any update to this issue?

wangguanzhi, 2020-04-03:
Hi @jackbart94!
I also want to know what could be the problem with this issue. My running server sometimes crashed when seg faults happened.

LinxiFan, 2020-04-04:
I'm also seeing the same issue, please help. Thanks!

AngelJA, 2020-07-11:
Curious if this has been fixed.  Also getting a segmentation fault after running a python script on version 0.9.9.

Stack trace points to ThreadPool.h:
carla::ThreadPool::Run (this=<optimized out>) at ../../LibCarla/cmake/../source/carla/ThreadPool.h:63

Is this related?

jackbart94, 2020-07-13:
Sorry for not updating but yes, it is already fixed in 0.9.9. This error was happening every time you had to shut down a python script involving Traffic Manager.

Not so sure what is happening in your script, maybe you could open a separate issue and give us more info!

pengyuan-zhou, 2021-01-27:
> Sorry for not updating but yes, it is already fixed in 0.9.9. This error was happening every time you had to shut down a python script involving Traffic Manager.
> 
> Not so sure what is happening in your script, maybe you could open a separate issue and give us more info!

Hi, the error still appears when running performance_benchmark.py