carla-simulator/carla #2405: Spawn new vehicles in front of existing vehicle

rjtshrm, 2020-01-27:
Hi, I am trying to spawn a new vehicle in front of existing vehicles. I tried something like below:

```
#get current vehicle transform to get both location & rotation
tf = current_v.get_tranform()
#Spawn new vehicle with some distance addition to above transformation location (say 10meter in x, y)
new_loc = tf.location + carla.Location(10, 10, 0)
world.spawn(new_vehicle, carla.Transform(new_loc, carla.Rotation())
```

I managed to spawn a new vehicle near to the existing one but not in front of it. It is actually random. Sometimes it is and sometimes not. 

There is a way I thought of is driving the existing vehicle a very small distance and see the change in the x-axis and y-axis whether it is increasing or decreasing and then decide whether to add or subtract. But is there any simple way of spawning cars in front of the vehicle always?

Also, the Location of spawning doesn't matter (it could be a road, park, garden) just random location.

fyj97, 2020-01-28:
Dear rjtshrm,
I have known your question. Because you don't know the heading angle of your ego-vehicle, the vehicle you generate seems random. You can use get_transform() to get the heading angle and do some simple calculations to do that.

rjtshrm, 2020-01-28:
Yes, Did this and working fine.

closed 2020-01-31 by rjtshrm

hamdii02, 2020-03-14:
@rjtshrm how you did this ?
could you please give your code
thanks