carla-simulator/carla #2411: Spawn walkers tm_spawn_npc.py -w 50

ll7, 2020-01-29:
Hello,

I am struggling to spawn 0 vehicles and 50 walkers.

Spawning 20 vehicles is working properly:
```
$ python3 tm_spawn_npc.py -n 20
Spawned 20 vehicles and 0 walkers, press Ctrl+C to exit.
```
Spawning 0 vehicles is working properly, spawning 50 walkers is **not**:
```
$ python3 tm_spawn_npc.py -n 0 -w 50
Spawned 0 vehicles and 0 walkers, press Ctrl+C to exit.
```
Spawning only 50 walkers is **not** working properly:
```
$ python3 tm_spawn_npc.py -w 50
Spawned 10 vehicles and 0 walkers, press Ctrl+C to exit.
```
I expected, that `-w` works similar to `-n` as it is implemented in: 
[Link to parameter implementation](https://github.com/carla-simulator/carla/blob/1f7669e4b308aa8e8f7612dde6666ba6eda14e9f/PythonAPI/examples/tm_spawn_npc.py#L47)

I am using:
* Ubuntu 18.04
* UE 4.22
* carla master branch

[Link to parameter implementation](https://github.com/carla-simulator/carla/blob/1f7669e4b308aa8e8f7612dde6666ba6eda14e9f/PythonAPI/examples/tm_spawn_npc.py#L47)

jackbart94, 2020-01-29:
Hi,

If you don't specify `-w` it still tries to spawn 50 walkers (see default parameter) so there is something going on when spawning walkers. I assume you haven't changed the script (if you have, double check please) but I suggest to try with `spawn_npc.py` as well. I tried to replicate the error and it worked just fine. If it's still wrong, provide as much information as possible so we can try to help you!

ll7, 2020-01-29:
Hey, 
thank you for the quick response. What do you mean by:
> I assume you haven't changed the script 
I assumed, that I pass the python script a parameter by typing `-w 50`. Therefore, I don't change the script, but specify how it's supposed to run?

ll7, 2020-01-29:
I forgot to `./Update.sh` between a branch swap. Sorry, that I bothered you.

closed 2020-01-29 by ll7