carla-simulator/carla #2416: Several errors while compiling 0.9.7 on Linux

exoticDFT, 2020-01-30:
I'm having issues compiling any of the available options for Carla 0.9.7.

**Carla version 0.9.7
Linux (Fedora 31)
Python 3.7.5
Unreal engine 4.22
Boost 1.69.0
Clang 9.0**

I'm writing the few commands I tried followed by the main error I'm getting directly following the command. The full outputs from the terminal for each command are attached as files, as they'll be too long to include here.

Building the Python API:
> `make PythonAPI`
>
> In file included from ../../LibCarla/source/carla/client/BlueprintLibrary.cpp:7:
../../LibCarla/cmake/../source/carla/client/BlueprintLibrary.h:40:5: error: explicitly defaulted move constructor is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
    BlueprintLibrary(BlueprintLibrary &&) = default;
    ^
> make: *** [Util/BuildTools/Linux.mk:117: LibCarla.client.release] Error 1

Building the C++ Carla Library:
> `make LibCarla`
>
> ../../LibCarla/source/test/common/test_streaming.cpp:58:3: error: reference to 'Server' is ambiguous
  Server<tcp::Server> srv(io.service, TESTING_PORT);
  ^
../../LibCarla/cmake/../source/carla/streaming/Server.h:20:9: note: candidate found by name lookup is 'carla::streaming::Server'
  class Server {
        ^
../../LibCarla/cmake/../source/carla/streaming/low_level/Server.h:25:9: note: candidate found by name lookup is 'carla::streaming::low_level::Server'
  class Server {
        ^
> make: *** [Util/BuildTools/Linux.mk:111: LibCarla.server.release] Error 1

Building the full package:
> `make package`
>
> ../../LibCarla/source/test/common/test_streaming.cpp:58:3: error: reference to 'Server' is ambiguous
  Server<tcp::Server> srv(io.service, TESTING_PORT);
  ^

This is actually the same as LibCarla, as I guess this is built first.

I realize that the team has focused on Ubuntu as the Linux distro and Clang/GNU 7 since it's the default available. I'm assuming these are only warnings in Clang/GNU 7. However, they seem to be breaking the c++ standard or places where the compiler may be doing something not expected by the team.

Would there be an easy way to ignore these errors? 

[compileErrorLibCarla.txt](https://github.com/carla-simulator/carla/files/4131784/compileErrorLibCarla.txt)
[compileErrorPackage.txt](https://github.com/carla-simulator/carla/files/4131785/compileErrorPackage.txt)
[compileErrorPythonAPI.txt](https://github.com/carla-simulator/carla/files/4131786/compileErrorPythonAPI.txt)

ben-z, 2020-03-27:
Did you manage to resolve this issue? I'm running into the same error when building on macOS with https://github.com/carla-simulator/carla/pull/2433

ncointe, 2020-04-14:
Same problem here, with Archlinux (rolling release 14th april) and Clang 9

Apparently there is an ambiguity issue in file LibCarla/source/test/common/test_streaming.cpp due to 
-the declaration of class Server in both carla/streaming/Server.h and carla/streaming/low_level/Server.h
-the declaration of class Server in both carla/streaming/Client.h and carla/streaming/low_level/Client.h

Solved with a few modifications :
Line 58 and 91, replace "Server" by "carla::streaming::low_level::Server"
Line 63 and 94, replace "Client" by "carla::streaming::low_level::Client"

closed 2020-06-20 by stale[bot]

qhaas, 2020-09-21:
Running into similar issues using clang8 while trying to build carla 0.9.8 with UE 4.23.1 (yes, unsupported, but we have a special use case).  I find their commit relating to the upgrade from clang7 to clang8 most helpful in dealing with these errors:  https://github.com/carla-simulator/carla/commit/457b63b85ef9227f36f7a3dee07da34c1c8d9276#diff-6fab226819b7750992e4832abd67a0a5

chjXu, 2022-06-21:
I have sovle this problem with add the namespace "carla::streaming::low_level::" before the Server and Client object in test_streaming.cpp file. Then, it's work to compile.